PIC-IRIS/PH5 #328: [BUG] ph5_validate bugs

nick-iris, 2019-08-12:
**Describe the bug**
`ph5_validate` does not correctly detect whether data exists outside of deploy/pickup times if the metadata contains multiple epochs.

**To Reproduce**
See experiment 19-013. In this experiment, stations 1001 through 1024 all have two epochs. 

For  this example, look at the first station 1001:

```
(ph5) -bash-4.2$ ph5tostationxml -n master.ph5 -p . --format=text --level=station --station=1001
#Network|Station|Location|Channel|Latitude|Longitude|Elevation|Depth|Azimuth|Dip|SensorDescription|Scale|ScaleFreq|ScaleUnits|SampleRate|StartTime|EndTime
6E|1001||EPZ|39.1988|-118.379|1585.26|0.0|0.0|90.0|Geospace GS11V/Reftek rt125|1627999798.08|50.0|M/S|200.0|2018-03-22T16:00:00|2018-03-23T01:00:00
6E|1001||EPZ|39.1975|-118.3825|1606.27|0.0|0.0|90.0|Geospace GS11V/Reftek rt125|1627999798.08|50.0|M/S|200.0|2018-03-23T16:00:00|2018-03-24T01:00:00
```

Now when `ph5_validate` is run against this experiment, it is marking station 1001 channel EPZ as having 86295 seconds of data after the pickup time:

```
Station 1001 Channel 1
0 error, 2 warning, 0 info
-=-=-=-=-=-=-=-=-
WARNING: No station description found.
WARNING: Data exists after pickup time: 86395 seconds
```

My suspicion is that the program is only comparing data availability against the first epoch. Anything before/after the first epoch, even if there is a gap in data availability is marked as being outside of the metadata deploy/pickup times.

**Expected behavior**

Whether data exists outside of deploy/pickup times needs to be determined by taking into consideration all station/channel epochs, not just the first epoch. Additionally some other warnings such as missing descriptions, should instead be labeled as `info` level messages. Ideally experiments sent to the DMC should have no warnings from `ph5_validate`.

**Additional context**
Experiments 16-006, 19-013, 19-014, and 19-004 all appear to have similar issues.

ascire-pic, 2019-08-13:
The ph5tostationxml example above is slightly misleading. The first epoch of station 1001 is part of array 1  (id_s 1001) and is correct. The second epoch of station 1001 is actually referencing array 2, id_s 2001, and the SEED station name should be 2001.

id_s 1001 in array 1 is DAS 12232. DAS 12232 has 2 epochs of data in the raw data file, epoch 1 on Mar 22 and epoch 2 on Mar 23. Epoch 1 is covered by array 1, id_s 1001. ph5_validate is complaining about the data from Epoch 2 which is described in array 2, id_s 2010.

[image]

dsentinel, 2019-08-13:
I think
https://github.com/PIC-IRIS/PH5/blob/9589838a0138cfca9f87e196ec2ca3a4a6ddb940/ph5/utilities/ph5validate.py#L484
should not happen here we can not validate all the data for a given sn on a station bases

Let's remove this check from this function.
For checking that all data is described we should loop over all the data by sn, and make sure there is one or more stations epochs that cover the data entirely

nick-iris, 2019-08-13:
> The ph5tostationxml example above is slightly misleading. The first epoch of station 1001 is part of array 1 (id_s 1001) and is correct. The second epoch of station 1001 is actually referencing array 2, id_s 2001, and the SEED station name should be 2001.

@ascire-pic Thanks for the explanation. Are you going to send us a new experiment 19-013 master.ph5 file  with the corrected SEED station names?

ascire-pic, 2019-08-13:
@nick-iris Yep, I put George on it. You should have a new master.ph5 for 19-013 soon.

dsentinel, 2019-08-15:
> I think
> https://github.com/PIC-IRIS/PH5/blob/9589838a0138cfca9f87e196ec2ca3a4a6ddb940/ph5/utilities/ph5validate.py#L484
>
>
> should not happen here we can not validate all the data for a given sn on a [...]

To further clarify. This will not be a problem for fairfield nodes as a redeployed node will be a different station and it seems nodes are indexed on station and not serial number. However all other input formats will have this issue. @derick-hess  may have further insight.
That said,  I still think iterating over the das table is the way to go. 
If runtime is an issue the station epochs to compare with can be saved as the stations are iterated over for validation.

damhuonglan, 2019-08-16:
@ascire-pic @nick-iris After I run ph5_validate in fix_328 for experiment 19-013, the following stations still have warning: Data exists after deploy time for the following stations: 1007, 1019,1021,1022,1023,1024  and the warning Data exists before deploy time for the following stations: 2009, 2011, 2016,2017,2019,2021. Please check if that is correct?

timronan, 2019-08-29:
PH5validate has the same bugs as PR #332 which both originate from the ph5api.get_extent function. As mentioned in PR # 314:

> ph5api.get_extent works by getting the start_times from the first and last DAS records. The end time is made by taking the start time of the last DAS record and adding (number of samples)/sample rate for that record. 

0r 
`latest_epoch = (latest_epoch_start +(float(new_das_t[-1]['sample_count_i'])/ true_sample_rate)`

The problem is that get_extent only looks for a das's time extent, which is non-unique to a specific array, station,  or epoch. We need a key,  maybe from inside the das record, to sort on so we get a unique array, station,  or epoch.  Example of das record:

```
{'time/micro_seconds_i': 0, 'sample_rate_i': 100, 'array_name_data_a': 'Data_a_0001', 'sample_count_i': 177000, 'stream_number_i': 1, 'time_table_n_i': 0, 'event_number_i': 1, 'array_name_SOH_a': 'SOH_a_0001', 'response_table_n_i': 0, 'array_name_log_a': '', 'sample_rate_multiplier_i': 1, 'time/epoch_l': 1458745200, 'time/type_s': 'BOTH', 'raw_file_name_s': 'I0836RAWRUDREMI.TRD', 'receiver_table_n_i': 0, 'array_name_event_a': '', 'time/ascii_s': 'Wed Mar 23 15:00:00 2016', 'channel_number_i': 1}
```
Currently, get-extent, has optional arguments for the start time and end time. These input arguments could be used to isolate specific epochs, and could potentially be fed into the get extent from the ph5 metadata.

gegert-pic, 2019-09-30:
I ran this branch on PH51, and I believe we can close this issue out. I have attached the before and after PH5_Validate Reports for good measure. 
[ph5_validate.log](https://github.com/PIC-IRIS/PH5/files/3673238/ph5_validate.log)
[old_ph5_validate.log](https://github.com/PIC-IRIS/PH5/files/3673239/old_ph5_validate.log)

commit e257e7c2 (2019-09-25): unittest for ph5validate.py for the change to fix #328 when there are multi_deploy time for the same das on one array
commit c80ba5af (2019-09-25): unittest for ph5validate.py for the change to fix #328 when there are multi_deploy time for the same das on one array
commit 0bd47ba5 (2020-05-22): incorrectly detect data between different deploy/pickup times (#357)

* not use read_das_t(), only use query_das_t() to consider component and sample_rate

* fix #328 ph5_validate does not detect data between different deploy/pickup times

* unitest for ph5validate.py for fixing #328 detect data between different deploy/pickup times

* in create_ph5.sh, add create testing PH5 file for ph5_validate #328

* metatadata to change array tables for testing issue #328

* Revert "in create_ph5.sh, add create testing PH5 file for ph5_validate #328"

This reverts commit 796029a7b1168e0a62ea6980b21ed731c [... cut]
closed 2019-09-30 by gegert-pic