PIC-IRIS/PH5 #329: [BUG] segdtoph5 did not create response table

ascire-pic, 2019-08-13:
segdtoph5 did not create a response table for a node experiment. Unlike other nodes where segdtoph5 correctly generated the response table, these nodes were programmed with a gain of 18 DB. All previously processed nodes were programmed with gain of 12 DB. 

Response table had to be created manually and added to the PH5. The Rutford Ice Stream nodes (where this problem was noted) are the first nodes programmed at PASSCAL with something other than 12 DB. Most nodes have been programmed with a gain of 12 DB so far. Other than Rutford Ice Stream, we do not currently have examples for the other possible gains.

Possible gains for node: 0 DB, 6 DB, 12 DB, 18 DB 24 DB, 30 DB, 36 DB
 
Server: ph52
segdtoph5 2019.14

A test PH5 was made using segdtoph5 via command line to create a PH5 with a single node. segdtoph5 did not create a response table in this test data set. The segd2ph5.log file (included here) does not indicate any problems. Also included is a copy of the commands run to make the test PH5. The test PH5 is located on ph52.

Location of files on ph52:
/ph52/Working/RutfordIceStream/no-resp-table

[from-command-line.txt](https://github.com/PIC-IRIS/PH5/files/3498417/from-command-line.txt)
[segd2ph5.log](https://github.com/PIC-IRIS/PH5/files/3498418/segd2ph5.log)

derick-hess, 2019-08-14:
That should be a quick fix. The original documentation for the nodes only described gains of 0, 12, 24, and 36.

Starting [here](https://github.com/PIC-IRIS/PH5/blob/9589838a0138cfca9f87e196ec2ca3a4a6ddb940/ph5/utilities/segd2ph5.py#L37)

Those other gains just need to be added in and things should work.

dsentinel, 2019-08-15:
Nice, where do the constant values come from?

derick-hess, 2019-08-16:
These just come from the equation dB= 20log(V1/V2)
[db to power ratio calculator](https://www.daycounter.com/Calculators/Decibels-Calculator.phtml)
The amplitude of the ADC is 2500 millivolt so that is 0dB
Looks like when Steve wrote this he rounded everything...
I calculate they should actually be be:

```python
LSB00 = 2500. / (2 ** 23)  # 0dB
LSB06 = 1250.  / (2 ** 23)   # 6dB
LSB12 = 628.14070 / (2 ** 23)  # 12dB
LSB18 = 314.86146 / (2 ** 23)  # 18dB
LSB24 = 158.22785 / (2 ** 23)  # 24dB
LSB30 = 79.11392 / (2 ** 23)  # 30dB
LSB36 = 39.61965 / (2 ** 23)  # 36dB = 39mV full scale
LSB = LSB36

LSB_MAP = {36: LSB36, 30:LSB30, 24: LSB24, 18:LSB18 12: LSB12, 6:LSB06, 0: LSB00}
```

damhuonglan, 2019-08-19:
@dsentinel : Will you fix this issue?

PR #333 (2019-09-13): segdtoph5 did not create response table, add gain of 18dB
fixes #329
closed 2019-09-13 by dsentinel