PIC-IRIS/PH5 #330: [BUG] NOVEN does not catch invalid sensor model name

akram-pic, 2019-08-16:
**Describe the bug**

NOVEN is specific on the format of sensor model. For example, it does not accept a dash in the text string describing a sensor model; L-22 should be L22 and L-4C should be L4C in the input csv for receiver array. 
However, NOVEN "check input" feature does not catch this. After loading a receiver array csv, assigning all column headers, and "checking input", it passes everything and gives the message of "Nothing funky found". But once trying to save out the file into kef, it complains about "can not continue, must open csv and check input". On the terminal, an error message like this shows up:
+++++++++++
Traceback (most recent call last):
  File "/home/field/PH5/ph5/utilities/noven.py", line 228, in run
    self.sep)
  File "/home/field/PH5/ph5/utilities/novenqc.py", line 493, in qc_receivers
    TABLE, err = mk_table(table, sep)
  File "/home/field/PH5/ph5/utilities/novenqc.py", line 144, in mk_table
    if t[0] == '#' or t == '':
IndexError: string index out of range_
+++++++++++

It is confusing that the error message is quite vague.

**Environment (please complete the following information):**
 - Program Version [noven version:2018.268]

**To Reproduce**
See the attached receiver array as an example. 
[noven-bug-sensmod-example-recarray.txt](https://github.com/PIC-IRIS/PH5/files/3510401/noven-bug-sensmod-example-recarray.txt)
 

**Expected behavior**

1- noven to be flexible in accepting the string format, e.g. all of these be accepted: L-22, L_22, L22
2- if above is not possible, noven to complain in "check input". 
and
3- check other columns for similar issues.

derick-hess, 2019-08-16:
I think this is actually an error with your input file having a blank line at the end.

nick-iris, 2019-08-16:
If what @derick-hess is saying about the blank line is in fact the issue, then this should be fixed. The program shouldn't crash because there is a blank line at the end of the csv file.

derick-hess, 2019-08-16:
Yeah it was on the fic list a while ago. I think Alissa ran into it once last year. It should be a fairly simple fix

akram-pic, 2019-08-16:
The original CSV file I ran noven on does not have an extra line at the end, the txt file I attached to github report has one as a result of creating it on my local machine.

akram-pic, 2019-08-16:
Checking with extra line at the end of the csv file with L-22, L-4C I still get the same error as mentioned above.

derick-hess, 2019-08-19:
I'll take a look at this. It is probably best anyway to automatically have it keep names consistent and remove the '-' if someone puts it in.

hrotman-pic, 2020-04-14:
A sensor name GS-11D was ok'd by noven, saved in a kef, and is still showing as GS-11D in the stationXML from PH5.  Does that mean this issue is resolved?

ascire-pic, 2020-10-23:
Tested again using same csv file where this issue was previously observed. Noven behaved as expected and I was able to create a kef file that I then loaded into a PH5 archive without any issues. Will close issue as it appears to have been resolved.

closed 2020-10-23 by ascire-pic