PIC-IRIS/PH5 #331: how does ph5 populate the datalogger "bit_weight/value_d"?

akram-pic, 2019-08-16:
This may or may not be a bug, but clarification is needed:

**Issue description:**
An experiment with only rt130 as datalogger is being created in ph5. The Response_t table has multiple Table rows with different bit_weight/value_d. Since Response_t is created before resp_load is run, I assume ph5 is pulling the bit_weight either from the raw rt130 logfile (?) or somewhere from the data dataheader(?). Checking the rt130 log files (for the day of data in ph5) for bit weight, I see discrepancies between logfile and Response_t (see below and attached files). 

Here is the content of Response_t (after resp_load) 
++++++++++++++++++++++++++++++++++++++++
_#	Thu Aug 15 15:03:47 2019	ph5 version: 4.1.2
#
_#   Table row 1__
/Experiment_g/Responses_g/Response_t
	n_i=0
	bit_weight/value_d=1.589
	bit_weight/units_s=uV/count
	gain/units_s=
	gain/value_i=1
	response_file_a=
	response_file_das_a=/Experiment_g/Responses_g/rt130_500_1_1
	response_file_sensor_a=/Experiment_g/Responses_g/L22

_#   Table row 2_
/Experiment_g/Responses_g/Response_t
	n_i=0
	bit_weight/value_d=1.589
	bit_weight/units_s=uV/count
	gain/units_s=
	gain/value_i=1
	response_file_a=
	response_file_das_a=/Experiment_g/Responses_g/rt130_500_1_1
	response_file_sensor_a=/Experiment_g/Responses_g/L4C

_#   Table row 3_
/Experiment_g/Responses_g/Response_t
	n_i=1
	bit_weight/value_d=1.591
	bit_weight/units_s=uV/count
	gain/units_s=
	gain/value_i=1
	response_file_a=
	response_file_das_a=/Experiment_g/Responses_g/rt130_500_1_1
	response_file_sensor_a=/Experiment_g/Responses_g/L22
++++++++++++++++++++++++++++++++++++++++

and checking the rt130 log file for "true_weight":

++++++++++++++++++++++++++++++++++++++++
_ES01 (D1ED): with L4C_

(ph5) [field@texserv2 test-log]$ more 2018.174.19.02.39.D1ED.log | grep "true weight" | uniq
true weights = 2.757 uV 2.757 uV 2.757 uV

_ES02 (929D): with L22_

ph5) [field@texserv2 test-log]$ more 2018.174.17.30.38.929D.log | grep "true weight" | uniq
true weights = 1.589 uV 1.591 uV 1.589 uV

_NW01 (AB2B): with L22_

(ph5) [field@texserv2 test-log]$ more 2018.174.14.12.39.AB2B.log | grep "true weight" | uniq
true weights = 1.586 uV 1.585 uV 1.585 uV

++++++++++++++++++++++++++++++++++++++++

note the large discrepancy in the example above: rt130-D1ED/L4C, response_t shows bit weight as 1.589 uV, while rt130 log is reporting 2.757 uV.

Now there are two questions: 

1- What is happening and where does ph5 pull the bit weight from, or if it calculates it, how is that working? 

2- With bit weight being a different value for each one of these rt130s, there is no indication of which of the rows in response_t belongs to which datalogger. What is the order? and how does ph5 assign the right response table row to the corresponding das table and rows in the array table?

----------------
Attachments:
three rt130 log files and response_t table (can provide raw data if needed)

[2018.174.17.30.38.929D.log](https://github.com/PIC-IRIS/PH5/files/3510994/2018.174.17.30.38.929D.log)

[2018.174.14.12.39.AB2B.log](https://github.com/PIC-IRIS/PH5/files/3510991/2018.174.14.12.39.AB2B.log)

[2018.174.19.02.39.D1ED.log](https://github.com/PIC-IRIS/PH5/files/3510995/2018.174.19.02.39.D1ED.log)

[response_t.kef-orig-back.txt](https://github.com/PIC-IRIS/PH5/files/3511000/response_t.kef-orig-back.txt)

derick-hess, 2019-08-19:
I'm not up on the rt130 code (its the exact same as in rt2ms and all other rt code we use) but it looks like it gets both the nominal and true bit weights from event headers. 

PH5 creates a response table for each sensor/das combo. The response_n_i number in the array table references the entry in the response table

derick-hess, 2019-08-19:
For anyone that wants to work on this. The pn130 code in ph5/core is the exact same as in rt2ms and other rt130 code. 1302ph5.py is where I think the issue probably is. That is the code that creates the tables, decides how to write the metadata data it is reading from the packets and loads the data.

ascire-pic, 2020-10-20:
Issue is being closed due to shifting development efforts to GeoHDF.

closed 2020-10-20 by ascire-pic