PIC-IRIS/PH5 #334: [BUG] indexing issue: response_t, das_t, and array_t reference incorrect "response_table_n_i"

akram-pic, 2019-08-20:
**Describe the bug**

Summary:
1- response_t assigns the same n_i=0 to two different sensor/das combo on two different Table row #. 
2- array_t and das_t reference to an incorrect "response_table_n_i" in the response_t. 
3- it is possible that as a result, this miss indexing does not allow proper creation of XML in response and channel level. 
   ph5tostationxml creates a XML in station level, but in response and channel level I get an error like this: "ph5.clients.ph5tostationxml - ERROR: A Code is" required

Details:
In this example I have three stations, two rt130/L22 and one rt130/L4C. response_t has three table rows, and assigns a n_i=0 to a L22 and L4C and n_i=1 to another L22. see the table summary below: 
 
response_t 

_#   Table row 1
/Experiment_g/Responses_g/Response_t
	n_i=0
das+sensor=rt130+L22	

_#   Table row 2
/Experiment_g/Responses_g/Response_t
	n_i=0
das+sensor=rt130+L4C

_#   Table row 3
/Experiment_g/Responses_g/Response_t
	n_i=1
das+sensor=rt130+L22

Extracting each of the das_t tables and looking at the response_table_n_i: Both das tables with L22 reference to both 0 and 1 in one das_t. For example, in one das_t_AB2B, I have response_table_n_i=0 and response_table_n_i=1 at the same time.
In the meantime, das_t_D1ED which is a L4C sensor, references response_table_n_i=1 as well. Note that from the response_t summary above, n_i=1 is not an L4C. 

Same indexing problem exists in array_t: response_table_n_i references to an incorrect table row in response table. 

I have hand edited each of the das_t, array_t and response_t to reference to the correct response_table_n_i, and loaded them all to master. However, I still get the code error in ph5tostationxml:

(ph5) [Sigma]$ ph5tostationxml -n master.ph5 -p /XXX/PH5-partial/Sigma -o XXX-partial.resp.xml --level=response
[2019-08-20 17:04:57,978] - ph5.clients.ph5tostationxml - ERROR: A Code is required

Despite the errors, I can extract miniseed data, with network code, name, time all correct. 
However still unable to create xml in response level. 

**Environment (please complete the following information):**

 - Program Version [pforma v2019.14], [ph5tokef 2018.268], [keftoph5 2019.036], [noven 2018.268] , [ph5tostationxml 2019.63]

**To Reproduce**
 See the original, unedited array, das and response tables attached. 
Note this the same project reported in the last issue report (#331): how does ph5 populate the datalogger "bit_weight/value_d"? 

**Expected behavior**

1- expect indexing to be correct
2- expect meaningful error messages in ph5tostationxml
3- expect ph5_validate to catch the error.

[indexing-bug-ph5.zip](https://github.com/PIC-IRIS/PH5/files/3521862/indexing-bug-ph5.zip)

akram-pic, 2019-09-04:
**additional note on indexing**

I came across a similar indexing issue as above, in an only node experiment: 

**Details:**
I have a data set comprising only of nodes. Looking at the array_t before resp_load, the "response_table_n_i " is set to "0" pointing to the correct n_i in the response table. 
However, after rep_load is applied, "response_table_n_i " changes to "1". This n_i does not exist in the response table. 
I examined other tables (das, receiver) and it doesn't seem like the indexing issue leaks into those tables. 
Manually editing the array table to the correct index and loading it back to ph5 fixed the errors in ph5_validate on not finding responses, and I was able to create the XML in response level as well.

**Summary**

looks like resp_load might be the reason behind some of these indexing issues. ( or perhaps some of the subroutines it references. We did not use to have any indexing problems back several months ago! )

dsentinel, 2019-10-23:
@damhuonglan if @nick-iris didn't address thin in #359, please take a look.

nick-iris, 2019-10-29:
I'm experiencing the same issue with another experiment. I think that `resp_load` may be trying to do too much.

It may be better to just have the program just load the RESP files and update the Response_t table. Then make it the users job to map the Response_t table entries to the Array_t table entries. Thoughts on this?

hrotman-pic, 2019-10-29:
All the PH5s I've built have needed a manual edit of the array table to fix response_table_n_i.  it would be wonderful to get this fixed.  Thanks for the high priority label.

dsentinel, 2019-11-22:
@damhuonglan,
Looks like you have solution going. Please create a PR for it.

damhuonglan, 2020-05-20:
Request 1 has been fixed and merge. Still need to work on ph5tostationxml and ph5validate before this issue can be closed.

closed 2021-05-18 by timronan