PIC-IRIS/PH5 #335: [BUG]segytoph5/segyreader glitches

dthomas1953, 2019-08-21:
**Describe the bug**
When running segytoph5 with a vetted segy file, execution crashes thusly:
```
$ segytoph5 -n master.ph5 -f data/Hansel_Valley_geo.segy
[2019-08-21 16:30:04,314] - ph5.utilities.segy2ph5 - INFO: segy2ph5 Version: 2018.268
[2019-08-21 16:30:04,314] - ph5.utilities.segy2ph5 - INFO: Opened: data/Hansel_Valley_geo.segy
[2019-08-21 16:30:04,314] - ph5.utilities.segy2ph5 - INFO: ['/Users/dthomas/miniconda/envs/ph5/bin/segytoph5', '-n', 'master.ph5', '-f', 'data/Hansel_Valley_geo.segy']
Traceback (most recent call last):
  File "/Users/dthomas/miniconda/envs/ph5/bin/segytoph5", line 11, in <module>
    load_entry_point('ph5', 'console_scripts', 'segytoph5')()
  File "/Users/dthomas/ph5/PH5/ph5/utilities/segy2ph5.py", line 1095, in main
    prof()
  File "/Users/dthomas/ph5/PH5/ph5/utilities/segy2ph5.py", line 1010, in prof
    th = SR.read_text_header()
  File "/Users/dthomas/ph5/PH5/ph5/core/segyreader.py", line 206, in read_text_header
    txt = ebcdic.EbcdicToAscii(eval(what))
  File "<string>", line 1, in <module>
NameError: name 'container' is not defined
Closing remaining open files:./master.ph5...done
```

**Environment (please complete the following information):**
 - OS: [e.g. Mac OSx 10.12.6 Sierra]
 - Program Version [segytoph5 2018.268, segyreader 2018.268]

**To Reproduce**
Have the segy file  Hansel_Valley_geo.segy at hand, try to use segytoph5 on it.

**Expected behavior**
It should have loaded the Hansel Valley segy file into the ph5 archive file.

**Additional context**
I have placed the Hansel Valley file here for further testing:
https://www.passcal.nmt.edu/~dthomas/bugfiles/Hansel_Valley_geo.segy

The problem occurs as segyreader is calling read_text_header in line 1010:
```python
def main():
    def prof():
        global INDEX_T_DAS, INDEX_T_MAP, RESP, EXREC, MINIPH5, ARRAY_T,\
            EVENT_T, FIRST_TIME, LAST_TIME, Das
        FIRST_TIME = sys.maxsize
        LAST_TIME = 0

        ARRAY_T = {}
        EVENT_T = {}

        MINIPH5 = None
        get_args()
        LOGGER.info("segy2ph5 Version: {0}".format(PROG_VERSION))
        LOGGER.info("Opened: {0}".format(SR.infile))
        LOGGER.info("{0}".format(repr(sys.argv)))
        initializeExperiment()
        RESP = Resp(EX.ph5_g_responses)
        rows, keys = EX.ph5_g_receivers.read_index()
        INDEX_T_DAS = Rows_Keys(rows, keys)
        rows, keys = EX.ph5_g_maps.read_index()
        INDEX_T_MAP = Rows_Keys(rows, keys)
        # Read text header
        th = SR.read_text_header()
```

During the execution of get_args(), SR is set up in line 184:

```python
    try:
        SIZE = os.path.getsize(args.infile)
        SR = segyreader.Reader(args.infile)
        SR.open_infile()
        if SR.FH is None:
            raise IOError()
    except Exception as e:
        LOGGER.error("Can't open infile (SEG-Y). {0}".format(e))
        sys.exit()
``` 

The actual problem occurs in segyreader.py.  I imported this, and segy_h.py, to creep into the bug in an interactive python shell.

```python
python
import segyreader, segy_h
SR = segyreader.Reader('data/Hansel_Valley_geo.segy')
SR.ext_hdr_type
'S'
SR.read_text_header()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "segyreader.py", line 208, in read_text_header
    txt = eval(what)
  File "<string>", line 1, in <module>
NameError: name 'container' is not defined
```

Note that we're getting the same ```NameError: name 'container' is not defined``` error.

It appears the error is happening between lines 194-208 of segyreader.py:
```python
    def read_text_header(self):
        ret = {}
        buf = self.read_buf(3200)
        t = segy_h.Text()

        t.parse(buf)

        keys = segy_h.Text().__keys__

        for k in keys:
            what = "container.{0}".format(k)
            if self.txt_hdr_type == 'E':
                txt = ebcdic.EbcdicToAscii(eval(what))
            else:
                txt = eval(what)
```

The buffer read works fine:
```python 
buf = SR.read_buf(3200)
```

And, the data are flowing in:
```
buf[3100:]
'\xc5\x1e\'*\xc5\x18X\xc1\xc5\x12\x08X\xc4\xbc\x13\x8f\xc4_\xc7z\xc4\x11\x98\x8cD*\x10!DPK\xb7D_\xeb"DYk0D?;\xd9D\x15\xbaX\xc4\x1c\xab4\xc4O\xd4\xdc\xc4zs\x9d\xc4\x93V)\xc4\x93,\x90\xc4v(8\xc4=\x01ZD\x13PNDr\xa0\xeeD\xd7$\xb8E\x13k\x0cE\x18}\xe5E\x1c3\xfb'```

Stepping thru the execution of lines 194-208 produces the following. We can see that "what" is a string variable of the form 'container._nn_'.
```python
t = segy_h.Text()
>>> tp = t.parse(buf)
>>> keys = t.__keys__
>>> for k in keys:
...     what = "container.{0}".format(k)
...     print what
...
container._01_
container._02_
. . .
container._39_
container._40_
```

The crash comes when we try to evaluate "what" in lines 206 or 208:
```python
            if self.txt_hdr_type == 'E':
                txt = ebcdic.EbcdicToAscii(eval(what))
            else:
                txt = eval(what)
```

This is not a valid form for the python eval() function:
```python
>>> what = 'container._35_'
>>> eval(what)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<string>", line 1, in <module>
NameError: name 'container' is not defined
```

Normally, eval operates on variables, not string constants:
```python
>>> x = 1
>>> eval('x+2')
3
```

It looks like segy_h has a constructor for the container variable, but that it's not being initialized properly during the call to segy_h.Text() (line 197 of segyreader.py), nor in the parsing of the buffer  t.parse(buf) (line 199).

dthomas1953, 2019-08-23:
**Some Progress, but a new Brick Wall**
The main() function at the end of segy_h.py showed how to write and read a segy file, and that works well. That alerted me to change the uses of .parse(buf) in 4 places in segyreader.py.

Here is one such:
```python
    def read_binary_header(self):
        buf = self.read_buf(400)
        b = segy_h.Reel(self.endianess)

        ret = {}
        container = b.parse(buf)  # dthomas added 'container = '
        ## old way was just: 
        ## b.parse(buf)

        keys = segy_h.Reel().__keys__
        for k in keys:
            what = "container.{0}".format(k)
            ret[k] = eval(what) . ## this eval() fails unless container is defined as above.

        return ret

```

Besides read_binary_header(self), the same error occurred in read_trace_header(self), read_extended_header(self), and read_text_header(self).

With these 4 corrections added to segyreader.py, execution now proceeds to the next step. But I've hit a new brick wall. In segd2ph5.py's main(), in prof(), it calls segyreader's `th = SR.read_text_header()` (line 1095), which in turn invokes `th = process_trace(th, bh, rh, eh, tr)` (line 1053), which calls `process_das()` (line 924), which hits an error in lines 866-875:
```
        tdoy = timedoy.TimeDOY(year=year,
                               month=None,
                               day=None,
                               hour=hour,
                               minute=minute,
                               second=seconds,
                               microsecond=0,
                               doy=doy,
                               epoch=None,
                               dtobject=None)
```

The problem here is that day of year doy (=Julian date) should be defined, and a valid date can be obtained from the year and the doy alone, but `doy = rh['day']` is used in line 862. 

Here's where it gets tricky. In segy2ph5.py's main(), line 1028, that's where rh[] is defined:
```python
            rh, eh, tr = read_trace()
```  
Over in read_trace() (starts on line 527) we have:
```
def read_trace():
    '''   Read SEG-Y trace header and trace data   '''
    if SR.isEOF():
        return None, None, None

    th = SR.read_trace_header()
    if not th:
        return None, None, None
    eh = SR.read_extended_header()
    tr = SR.read_trace(SR.samples_per_trace, SR.bytes_per_sample)
    # Trace header, extended trace header, trace
    return th, eh, tr
```

It turns out that the header rh[] is defined as `SR.read_trace_header()` (the first argument read_trace() returns is th, which maps to rh[] in segy2ph5.py's main(), line 1028.

Here is the new sequence of errors, with the 4 container bugs fixed:

```
(ph5) spike:PH5 dthomas$ segytoph5 -f ../17-007_Hansel_Valley/data/Hansel_Valley_geo.segy -n ../17-007_Hansel_Valley/master.ph5
[2019-08-23 20:39:12,144] - ph5.utilities.segy2ph5 - INFO: segy2ph5 Version: 2018.268
[2019-08-23 20:39:12,145] - ph5.utilities.segy2ph5 - INFO: Opened: ../17-007_Hansel_Valley/data/Hansel_Valley_geo.segy
[2019-08-23 20:39:12,145] - ph5.utilities.segy2ph5 - INFO: ['/Users/dthomas/miniconda/envs/ph5/bin/segytoph5', '-f', '../17-007_Hansel_Valley/data/Hansel_Valley_geo.segy', '-n', '../17-007_Hansel_Valley/master.ph5']
[... 6 lines cut ...]
    prof()
  File "/Users/dthomas/ph5/PH5/ph5/utilities/segy2ph5.py", line 1053, in prof
    th = process_trace(th, bh, rh, eh, tr)
  File "/Users/dthomas/ph5/PH5/ph5/utilities/segy2ph5.py", line 924, in process_trace
    process_das()
  File "/Users/dthomas/ph5/PH5/ph5/utilities/segy2ph5.py", line 875, in process_das
    dtobject=None)
  File "/Users/dthomas/ph5/PH5/ph5/core/timedoy.py", line 87, in __init__
    month, day = self.getMonthDay(year, doy)
  File "/Users/dthomas/ph5/PH5/ph5/core/timedoy.py", line 168, in getMonthDay
    "Value for month, {0}, out of range!".format(month))
ph5.core.timedoy.TimeError: Value for month, 0, out of range!
Closing remaining open files:./../17-007_Hansel_Valley/master.ph5...done./miniPH5_00001.ph5...done
```

And here are the contents of rh[], as of line 860 (inside process_das()):
```
rh: {'recUpholeTime': 0, 'traceID': 1, 'hour': 0, 'sweepEnd': 0, 'sweepTaperAtStart': 0, 'lineSeq': 1, 'sourceToRecDist': -171, 'dataUse': 0, 'weatheringVelocity': 0, 'hiCutFreq': 0, 'gainType': 0, 'second': 0, 'totalStatic': 0, 'sourceSurfaceElevation': 12975330, 'sourceStaticCor': 0, 'sweepType': 0, 'taperType': 0, 'notchFreq': 0, 'phoneRollPos1': 0, 'muteStart': 0, 'traceWeightingFactor': 0, 'recLongOrX': 35860920, 'vertSum': 0, 'sweepLength': 0, 'phoneFirstTrace': 0, 'sourceLongOrX': 35845854, 'horSum': 1, 'taperOvertravel': 0, 'delay': 0, 'recLatOrY': 461803596, 'recElevation': 13053781, 'sweepTaperAtEnd': 0, 'minute': 0, 'sourceWaterDepth': 0, 'recStaticCor': 0, 'sourceUpholeTime': 0, 'lagTimeB': 0, 'muteEnd': 0, 'coordUnits': 3, 'elevationScale': -10000, 'notchSlope': 0, 'sweepStart': 0, 'recWaterDepth': 0, 'subWeatheringVelocity': 0, 'aliasFreq': 417, 'coordScale': -100, 'traceInEnsemble': 0, 'lagTimeA': 0, 'datumElevRec': 0, 'sourceDepth': 0, 'initialGain': 0, 'year': 12, 'event_number': 100, 'day': 0, 'sampleLength': 1501, 'correlated': 0, 'gainConst': 0, 'hiCutSlope': 0, 'sourceLatOrY': 461811584, 'deltaSample': 1000, 'energySourcePt': 100, 'reelSeq': 1, 'lowCutFreq': 0, 'aliasSlope': 0, 'timeBasisCode': 0, 'channel_number': 2, 'phoneLastTrace': 0, 'datumElevSource': 0, 'gapSize': 0, 'lowCutSlope': 0, 'cdpEns': 37}
```

We can see that year is defined as 12 (?!?), day as 0, and month and doy are not defined at all.

Some background: segy2ph5.py was started by Steve Azevedo, but never was finished. Alissa says someone has been hired at the DMC to process old segy files into ph5 archives, and that's why this unfinished project is needing some attention.

This is as far as I can get for now, help is requested.

dthomas1953, 2019-08-23:
I'll make a new PR related to this issue. Stand by.

timronan, 2019-08-29:
@dthomas1953 we created a branch, segytoph5, and rewrote the segy2ph5 function. We didn't use the old function as a starting point, so the new function is likely very different. @nick-iris and I are doing further testing and will make a pull request soon. Please feel free to pull down this branch and test out the new function. It would be good to get some feed back from multiple alpha testers. Please expect a pull request soon.

dthomas1953, 2019-08-30:
@timronan I tried it on my segy file, didn't get too far.
```
(ph5) spike:17-007_Hansel_Valley dthomas$ segytoph5 -n master.ph5 --segydir data
/Users/dthomas/miniconda/envs/ph5/lib/python2.7/site-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
Traceback (most recent call last):
  File "/Users/dthomas/miniconda/envs/ph5/bin/segytoph5", line 11, in <module>
    load_entry_point('ph5', 'console_scripts', 'segytoph5')()
  File "/Users/dthomas/ph5/PH5_copy/ph5/utilities/segy2ph5.py", line 144, in main
    if pass_SR is False:
UnboundLocalError: local variable 'pass_SR' referenced before assignment
Closing remaining open files:./master.ph5...done
```

data/ has the file Hansel_Valley_geo.segy. I tried adding that to the segydir, to no avail.