PIC-IRIS/PH5 #343: Clean up codebase.

dsentinel, 2019-09-04:
```
ph5/core/segyfactory.py
ph5/core/sfactory.py
```
sfactory seems to be an older version of segfactory
There seems to be more of these throughout the package.

dthomas1953, 2019-09-04:
There are. This one is vestigial as well: `/ph5/core/cs2cs.py`
It is called only by segyfactory.py and sfactory.py.

dsentinel, 2019-09-06:
`ph5/core/rt_130_h.pyc`
and all the .so's like
```
bcd_py.so  firfilt_py.so  ibm2ieee_py.so  rt_125a_py.so  rt_130_py.so
```
that are in the root dir

nick-iris, 2019-09-06:
@dsentinel The `.so` files are created when the c module extensions are created by setup tools.
i.e. setup.py

```python
    # install c-dependencies
    ext_modules=[Extension(*subcd_py.get_extension_options(),
                           include_dirs=[numpy.get_include()]),
                 Extension(*sufirfilt_py.get_extension_options(),
                           include_dirs=[numpy.get_include()]),
                 Extension(*suibm2ieee_py.get_extension_options(),
                           include_dirs=[numpy.get_include()]),
                 Extension(*surt_125a_py.get_extension_options(),
                           include_dirs=[numpy.get_include()]),
                 Extension(*surt_130_py.get_extension_options(),
                           include_dirs=[numpy.get_include()])]
``` 

They should be added to `.gitignore` if they aren't already.

dsentinel, 2019-09-06:
They are under revision control:
```
(ph5) ph5@ph5-2:/home/ph5/PH5$ git log  --oneline bcd_py.so
9f5caa8 more clean up
3d86339 first commit
```

nick-iris, 2019-09-06:
Oh wow those definitely should be removed then. Good call.

dsentinel, 2019-09-06:
Apologies... the `.so`'s were in git at one point in history.
They are not currently.

closed 2020-02-20 by nick-iris