PIC-IRIS/PH5 #344: [FEATURE] ph5_validate flag to suppress non-critical output

ascire-pic, 2019-09-05:
**Is your feature request related to a problem? Please describe.**
ph5_validate outputs a warning for certain non-critical things, like missing station description. Since many array tables in PH5 do not contain a station description, these warnings in particular can be overwhelming. These warnings are useful, but can make it harder to find critical issues.

**Describe the solution you'd like**
A flag to only display critical errors  (for example, missing data, missing responses, data outside of deploy/pickup times). Essentially a way to only see errors that could be critical from the perspective of having a valid archive.

**Additional context**
ph5_validate gave 1441 warnings about "no station description found" for the SHIRE PH5. Since the warnings or errors are on a separate line from the station/channel information, there did not seem to be any other way to find actual problems other than scrolling through the entire ph5_validate.log file.

derick-hess, 2019-09-05:
You can use the -l or --level flag to set the output level.  Options are either ERROR, WARNING, or INFO. 

INFO will give you everything
WARNING will give you warnings and errors
ERROR should give you only things that make the archive invalid

ascire-pic, 2019-09-05:
Somehow I missed that those flags are already there. Thank you!

closed 2019-09-05 by ascire-pic