PIC-IRIS/PH5 #348: Turkey hunting, remove unused code.

dsentinel, 2019-09-13:
There are unused modules, cli's, and functions. Some don't even run. Some create dependencies that may no longer be needed.

There are some functions written to handle specific experiments/datasets. These should be dropped, made for general use, or places somewhere appropriate.

The low hanging fruit is to remove the modules; files/cli tools/entry-points that are not being used, and dependencies unique to them.

The DG at PIC is going to compile a list of CL tools (entry points) that they don't use.
When that reaches a consensus the list can be posted here and the axing can begin.

Ideally, this should happen before unit tests get expanded like #347 
We don't want to right test for code that's going to disappear.

dsentinel, 2019-09-13:
The turkey: https://github.com/PIC-IRIS/PH5/blob/9cf2ad34e0949d68d02d749b8ffb760b5816aebe/ph5/core/mplturkey.py#L74

ascire-pic, 2020-10-20:
Issue is being closed due to shifting development efforts to GeoHDF.