PIC-IRIS/PH5 #353: [BUG] resp_load doesn't create response_t.kef

nick-iris, 2019-09-20:
**Describe the bug**
Running `resp_load` loads the .RESP files but fails to make a `response_t.kef` file. 

**Expected behavior**
Running `resp_load` should produce an updated `array_t.kef` file and a `response_t.kef` file. This program needs a lot of work. The interface is counter intuitive and the program makes some potentially destructive actions.

dsentinel, 2019-10-23:
Does this address the problems described in #334?

hrotman-pic, 2019-10-25:
Hi Nick, resp_load creates an updated response table that can be dumped out.  If you get stdout like this:
[2019-10-23 15:00:00,429] - ph5.utilities.resp_load - INFO: response_t.kef written into PH5
[2019-10-23 15:00:06,549] - ph5.utilities.nuke_table - INFO: Writing table backup: Array_t_001_2019023_00.kef.
[2019-10-23 15:00:06,554] - ph5.utilities.nuke_table - INFO: 1 It worked.
[2019-10-23 15:00:06,893] - ph5.utilities.nuke_table - INFO: Writing table backup: Array_t_002_2019023_00.kef.
[2019-10-23 15:00:06,896] - ph5.utilities.nuke_table - INFO: 2 It worked.
[2019-10-23 15:00:07,256] - ph5.utilities.kef2ph5 - INFO: Loading array_t_1.kef into master.ph5.
[2019-10-23 15:00:07,335] - ph5.utilities.kef2ph5 - INFO: Updated kef2ph5.log file.
[2019-10-23 15:00:07,680] - ph5.utilities.kef2ph5 - INFO: Loading array_t_2.kef into master.ph5.
[2019-10-23 15:00:07,702] - ph5.utilities.kef2ph5 - INFO: Updated kef2ph5.log file.
[2019-10-23 15:00:08,848] - ph5.utilities.nuke_table - INFO: Writing table backup: Array_t_001_2019023_01.kef.
[2019-10-23 15:00:08,854] - ph5.utilities.nuke_table - INFO: 1 It worked.
[2019-10-23 15:00:10,220] - ph5.utilities.kef2ph5 - INFO: Loading array_t_1.kef into master.ph5.
[2019-10-23 15:00:10,277] - ph5.utilities.kef2ph5 - INFO: Updated kef2ph5.log file.
[2019-10-23 15:00:10,803] - ph5.utilities.resp_load - INFO: array_t_1 Loaded into PH5
[2019-10-23 15:00:11,163] - ph5.utilities.nuke_table - INFO: Writing table backup: Array_t_002_2019023_01.kef.
[2019-10-23 15:00:11,167] - ph5.utilities.nuke_table - INFO: 2 It worked.
[2019-10-23 15:00:12,520] - ph5.utilities.kef2ph5 - INFO: Loading array_t_2.kef into master.ph5.
[2019-10-23 15:00:12,543] - ph5.utilities.kef2ph5 - INFO: Updated kef2ph5.log file.
[2019-10-23 15:00:13,069] - ph5.utilities.resp_load - INFO: array_t_2 Loaded into PH5

The backup tables are the unmodified ones.  The arrays loaded into PH5 are the updated ones.  It would be nice if it wrote out a separate response.kef file, but otherwise I've generally found it works as expected.

nick-iris, 2019-10-25:
I've been working on improving this program on PR #359. The program automatically updates the response table and the updates the arrays to reference the correct response. Before I made some changes I was running into problems loading response files.

hrotman-pic, 2019-10-25:
I saw you mention in the PR that it didn't work at all for you previously.  Did you mean the input.csv template wouldn't populate, or that it wouldn't load the response files with the -i input.csv option?

nick-iris, 2019-10-25:
I was running into a case where the program would fail and return a stack trace. Nonetheless, I think the changes I made are definitely an improvement. Among some small changes, I updated the code to use the Kefx library instead of making some dangerous assumptions about kef files.

nick-iris, 2020-02-20:
Fixed in https://github.com/PIC-IRIS/PH5/pull/373

closed 2020-02-20 by nick-iris