PIC-IRIS/PH5 #369: [BUG] stationXML shows different times on network and channel level

akram-pic, 2019-11-12:
**Describe the bug**

A node experiment was archived using a developmental branch on PIC server ph51 (see below). Upon finishing the archiving, ph5tostationxml outputs different times for network and channel level.

This experiment was a rolling node experiment, where nodes were pulled out of a location and replaced with a new node at the same location occupying the same station ID. That is the reason a developmental branch was used for archiving.  

(ph5) -bash-4.2$ ph5tostationxml -n master.ph5 --level network --format text 
#Network|Description|StartTime|EndTime|TotalStations
X5|Bedload_Monitoring|2019-06-29T14:51:01|2019-08-31T17:08:46|72

(ph5) -bash-4.2$ ph5tostationxml -n master.ph5 --level channel --format text --starttime=2019-08-31T17:08:46 --endtime=2019-10-01T00:00:00.0
#Network|Station|Location|Channel|Latitude|Longitude|Elevation|Depth|Azimuth|Dip|SensorDescription|Scale|ScaleFreq|ScaleUnits|SampleRate|StartTime|EndTime
X5|1111||GP1|34.092164|-106.869622|1408.0|0.0|0.0|0.0|Geo Space GS-30CT/FairfieldNodal ZLAND 3C|1029640000.0|40.0|M/S|1000.0|2019-08-31T16:33:56|2019-09-28T14:30:37
X5|1111||GP2|34.092164|-106.869622|1408.0|0.0|90.0|0.0|Geo Space GS-30CT/FairfieldNodal ZLAND 3C|1029640000.0|40.0|M/S|1000.0|2019-08-31T16:33:56|2019-09-28T14:30:37

**Environment (please complete the following information):**
++++++++++++

ph5-1:/storage/PH5$ git status
On branch Arrays_of_multideployment
Your branch is up to date with 'origin/Arrays_of_multideployment'.
++++++++++++
 

**To Reproduce**
See the full archive on ph51: ../../completed/19-010_Bedload_Monitoring

**Expected behavior**
network level to report start/end time of the first starting channel and last running channel to capture the whole time window of the experiment.

hrotman-pic, 2019-11-13:
I encountered this on another node experiment.  This was not a rolling experiment, but we updated the array tables later to capture some changes in station azimuth.

Original network bounds, before the array tables were updated:
<Created>2019-11-06T15:45:25.178064</Created>
  <Network code="1B" endDate="2019-08-19T20:43:12" startDate="2019-08-01T17:10:04"

Current network bounds, after the updated array tables were loaded:
<Created>2019-11-13T15:12:05.584642</Created>
  <Network code="1B" endDate="2019-08-14T18:04:00" startDate="2019-08-01T17:10:04"

The 2019-08-14T18:04:00 end date matches the end of the first epoch of the last station that was deployed.

This archive was constructed on ph51, but no branch was selected.

damhuonglan, 2020-05-14:
close after PR#371 was merged.

closed 2020-05-14 by damhuonglan