PIC-IRIS/PH5 #374: [BUG] ph5toms writes only one component

tjnewton, 2020-01-28:
**Describe the bug**
ph5toms does not write miniseed files for multiple components, even if the desired components are specified with the -c flag. 

**Environment**
 - OS: macOS 10.15.2
 - Program Version: PH5 4.1.3 (latest)

**To Reproduce**
`ph5toms -n master.ph5 -s 2018-03-13T00:00:00.00 -t 2018-03-13T23:59:59.99 -o mseed -F MSEED -c 1,2,3 --station_id 1`

**Expected behavior**
Should write mseed files for the three specified components 1, 2, and 3, or DP1, DP2, and DPZ. Instead a miniseed file for only the first component in the list is generated. If the -c flag is not included ph5toms writes only the first component. The behavior is the same for the --channel flag, and is independent of syntax, e.g., "-c 1,2,3" "-c=1,2,3" "-c DP1,DP2,DPZ" "--channel 1,2,3" "--channel=1,2,3" " --channel DP1,DP2,DPZ" all produce the same undesired result.

nick-iris, 2020-02-06:
I also have observed this behavior in the `ph5toms` client. This needs to be fixed ASAP.

closed 2020-02-07 by nick-iris

tjnewton, 2020-02-20:
Thanks Nick!

PR #377 (2020-02-07): Fix #374 - ph5toms now writes all requested components
### What does this PR do?
Fix #374 - `ph5toms` was only writing one component. This is a fix to make the program correctly write all requested channels.

### Relevant Issues?
#374 

### Checklist
- [x] This PR is not directly related to an existing issue (which has no PR yet).
- [x] All tests pass.
- [x] Any new or changed features have are documented.
- [x] Changes have been added to `CHANGELOG.txt` .
- [x] First time contributors have added your name to `CONTRIBUTORS.txt` .
commit 9f7a3813 (2020-02-07): Fix #374 - ph5toms now writes all requested components (#377)

Fix #374 - ph5toms now writes all requested components