PIC-IRIS/PH5 #380: [BUG] PySide missing on installation of Ubuntu 18.04

aringler-usgs, 2020-02-10:
**Describe the bug**
PH5 not installing on Ubuntu 18.04 because of PySide missing

**Environment (please complete the following information):**
 - OS: 18.04
 - Program Version V4.1.2

**To Reproduce**
Steps to reproduce the behavior:

```
(ph5) aringler@igskgacggsresch:~/inst_test/smart_solo/PH5$ pforma
Traceback (most recent call last):
  File "/home/aringler/anaconda3/envs/ph5/bin/pforma", line 11, in <module>
    load_entry_point('ph5', 'gui_scripts', 'pforma')()
  File "/home/aringler/anaconda3/envs/ph5/lib/python2.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/aringler/anaconda3/envs/ph5/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/home/aringler/anaconda3/envs/ph5/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/home/aringler/anaconda3/envs/ph5/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/aringler/inst_test/smart_solo/PH5/ph5/utilities/pformagui.py", line 10, in <module>
    from ph5.core import pmonitor
  File "/home/aringler/inst_test/smart_solo/PH5/ph5/core/pmonitor.py", line 37, in <module>
    raise ImportError(msg)
ImportError: 

No module named PySide. Please install PySide first, it is needed for pmonitor. 
```
After installing PySide ph5 can still not find PySide even thought it is installed:
```
(ph5) aringler@igskgacggsresch:~/inst_test/smart_solo/PH5$ python
Python 2.7.14 | packaged by conda-forge | (default, Dec  9 2017, 16:18:43) 
[GCC 4.8.2 20140120 (Red Hat 4.8.2-15)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import PySide
>>> 
```

**Expected behavior**

I get a similar issue on my mac, but I am able to resolve this by installing PySide manually.  

**Additional context**

@nick-iris was also able to replicate this behavior but provided a fix for the mac.

dsentinel, 2020-02-12:
@aringler-usgs, Sorry to hear you're having issues. Are you still in need of this or did the other method I emailed get you sorted?

aringler-usgs, 2020-02-12:
I was able to install this via a mac.  I will give the other method a try later today.  Thanks for that!

I put this ticket in as more of an FYI, in case other folks are having this issue.

Thanks!

dsentinel, 2020-02-12:
Great, I'm glad you're not held up.
We'll leave this open for information's sake.

closed 2020-02-20 by dsentinel