PIC-IRIS/PH5 #381: pforma not loading some fcnt files

jchaput82, 2020-02-26:
I am loading faifield nodal data into pforma (fcnt). Several of the files fail immediately to load, returning the error "assigned temporarily unavailable resource". Is this related to assigning a computer core that doesn't exist? I'm befuddled.
Cheers

Julien Chaput

dsentinel, 2020-03-10:
@jchaput82 , Did you get this sorted?
If you're still having issues, can you provide one or 2 culprit files?

jchaput82, 2020-03-11:
Sort of; I still don't know why proforma refused to load some of the files,
but I suspect it was a communication problem between it and the computer (I
think pforma tried to use a core that was not available or something).
Loading the "problem stations" as a subset worked fine, as did command line
tools.
A bigger problem is the current build. I've now tried on 3 different
platforms, and run into problems related to Pyside (used by pforma in the
GUI). I thankfully have access to an older build that works, but I think
the current one is broken.
Cheers

Julien

ascire-pic, 2020-10-20:
Based on the previous comment, it sounds like the original issue with pforma not loading some fcnt files has been worked around, if not solved. Based on this, I'm closing this issue. 

If problems with the newest build occur, a new issue should be opened.

closed 2020-10-20 by ascire-pic