PIC-IRIS/PH5 #386: [BUG] PH5 library is modifying sample_rate_multiplier value

nick-falco, 2020-03-09:
**Describe the bug**
The PH5 library is changing the value of the sample_rate_multiplier_i in the ph5/core/experiment.py module.

See https://github.com/PIC-IRIS/PH5/blob/master/ph5/core/experiment.py#L696-L700

```python
        # We look through each Das_t line and make sure it has the column
        # sample_rate_multiplier_i and if it is missing we set it to 1.
        # This column was added after data had been archived without it
        # at the DMC.
        for r in ret_read:
            if 'sample_rate_multiplier_i' not in r:
                r['sample_rate_multiplier_i'] = 1
            elif r['sample_rate_multiplier_i'] == 0:
                r['sample_rate_multiplier_i'] = 1

```
**Expected behavior**
I don't think the experiment.py module should be changing the value of data.  If the sample rate multiplier is missing from the dataset, then an error should be raised or the reader should just omit this value. Also if the sample rate multiplier is 0 in the dataset, then 0 should be returned when the Das_t table is read.

There are currently problems where `ph5availability` indicates that there is data available, but `ph5toms` returns no data. This issue stems from the sample_rate_multiplier_i value of 0.

For example, see experiment 09-005:
```
ph5toms -n master.ph5 -p . --station 7210 -s 2008-10-31T00:00:00.000000 -t 2008-10-31T23:59:59.999999
```
Returns No Data.

While
```
(ph5) -bash-4.2$ ph5availability -n master.ph5 -p . -a 2 --station 7210
#n s     l  c   q                    earliest                      latest
Z9 7210  -- DPZ   2008-10-31T04:00:00.000000Z 2008-10-31T04:29:59.888000Z
Z9 7210  -- DPZ   2008-11-02T01:58:00.000000Z 2008-11-02T03:59:59.888000Z
Z9 7210  -- DPZ   2008-11-02T04:58:00.000000Z 2008-11-02T05:14:59.888000Z
Z9 7210  -- DPZ   2008-11-02T05:58:00.000000Z 2008-11-02T06:14:59.888000Z
Z9 7210  -- DPZ   2008-11-02T06:58:00.000000Z 2008-11-02T07:14:59.888000Z
Z9 7210  -- DPZ   2008-11-02T07:58:00.000000Z 2008-11-02T08:14:59.888000Z
```
shows that there is data available.

Note that when you run
```
ph5tokef -n master.ph5 -p . -D 11149 
```
The kef file shows a sample_rate_multiplier_i value of 1, but in reality the value in the dataset is 0. This is misleading.

dsentinel, 2020-03-10:
I agree that this seems dodgy; overwriting the multiplier on a read... Eep!

There must be 2 code paths here. Does `ph5toms` not use `experiment.read_das()` or `ph5api.read_das_t()` which calls the former?

Where/how is `ph5toms` failing?
What is the SR for 
```
Z9 7210  -- DPZ   2008-10-31T04:00:00.000000Z 2008-10-31T04:29:59.888000Z
```

I would think that this is some kind of historical hack.
I imagine the `multiplier` here is the same as SEED. In SEED multiplier of 0 is undefined. So PH5 shouldn't be creating and `0`'s in this field, and if they exist, should probably be interpreted as a `1`, or `raise` as you stated.
Are we creating PH5's with multipliers of `0` that should be `1`? Looks like you have an example, we should probably track this code down and fix the heart of the problem, and update the archive or continue to assume `0` means `1`.

hrotman-pic, 2020-03-10:
Here's a snippet of a das table on one of the affected experiments, sample_rate_multiplier_i isn't present:
#
#       Fri Mar  6 18:37:15 2020        ph5 version: 4.1.2
#
#   Table row 1
/Experiment_g/Receivers_g/Das_g_11554/Das_t
        array_name_SOH_a=SOH_a_0001
        array_name_data_a=Data_a_0001
        array_name_event_a=
        array_name_log_a=
        channel_number_i=2
        event_number_i=1
        raw_file_name_s=
        receiver_table_n_i=0
        response_table_n_i=0
        sample_count_i=900000
        sample_rate_i=250
        stream_number_i=1
        time/ascii_s=Mon Sep  8 17:45:00 2008
        time/epoch_l=1220895900
        time/micro_seconds_i=0
        time/type_s=BOTH
        time_table_n_i=0

in ph51/Working/08-020-toFix

For this one, I've checked the array tables and the sample_rate_multiplier_i is present and correctly set to 1.

So we need to add a column to the das table?  I've only really used :Update with fix_3chan_texans to modify das tables, not added a column.

nick-falco, 2020-03-10:
The ph5ws-dataselect service code returns with the following stack trace:

```
Traceback (most recent call last):
  File "run_ph5dataselect.py", line 9, in <module>
    main()
  File "run_ph5dataselect.py", line 5, in main
    ph5dataselect.run_dataselect_service()
  File "/Users/nick/Development/workspace/pic-ph5/PH5WebService/PH5DataQuery/ph5dataselect.py", line 659, in run_dataselect_service
    write_to_stream(data_format, request_dict)
  File "/Users/nick/Development/workspace/pic-ph5/PH5WebService/PH5DataQuery/ph5dataselect.py", line 636, in write_to_stream
    wrote_data = write_mseed(request_dict)
  File "/Users/nick/Development/workspace/pic-ph5/PH5WebService/PH5DataQuery/ph5dataselect.py", line 341, in write_mseed
    for stream in data_obj.process_all():
  File "/Users/nick/Development/workspace/PH5/ph5/clients/ph5toms.py", line 1011, in process_all
    stream = self.create_trace(cut)
  File "/Users/nick/Development/workspace/PH5/ph5/clients/ph5toms.py", line 548, in create_trace
    apply_time_correction=nt, das_t=das)
  File "/Users/nick/Development/workspace/PH5/ph5/core/ph5api.py", line 1172, in cut
    float(d['sample_rate_multiplier_i'])
ZeroDivisionError: float division by zero
```

There error occurs in`ph5api`:
https://github.com/PIC-IRIS/PH5/blob/master/ph5/core/ph5api.py#L1172-L1174

The das table is read using the `filter_das_t` method:
https://github.com/PIC-IRIS/PH5/blob/master/ph5/core/ph5api.py#L1124

We could put in a special case to handle the missing or zero `sample_rate_multiplier_i` value, but I kind of feel that this should be fixed in the data if possible.

dsentinel, 2020-03-10:
Does this mean there are 2 code paths? ... There are also another sample_rate and multiplier in Sorts_g/Array_t_xxx as well as Receivers_g/Das_g_xxx/Das_t

I think the data should definitely be fixed, and that the code should default to 1 and not allow 0.

hrotman-pic, 2020-03-11:
> Does this mean there are 2 code paths? ... There are also another sample_rate and multiplier in Sorts_g/Array_t_xxx as well as Receivers_g/Das_g_xxx/Das_t

More or less.  I think we talked about this in a standup once.  response_table_n_i and receiver_table_n_i are also in both the das and array tables, but don't necessarily match and the array table value 'wins'.  For the snippet I gave above, that particular DAS is the DP1 component of a 3-channel Texan, and in the array table receiver_table_n_i=1 (doesn't match) and response_table_n_i=0 (does match).  There are other examples of this from more recent experiments.

nick-falco, 2020-03-11:
> Does this mean there are 2 code paths? ... There are also another sample_rate and multiplier in Sorts_g/Array_t_xxx as well as Receivers_g/Das_g_xxx/Das_t

From what I can tell, there are two code paths in `ph5api` for reading the das table:

`ph5api.query_das_t` : : https://github.com/PIC-IRIS/PH5/blob/master/ph5/core/ph5api.py#L765

`ph5api.filter_das_t` : : https://github.com/PIC-IRIS/PH5/blob/master/ph5/core/ph5api.py#L1885

I think part of the problem is that the `ph5api.query_das_t` method accesses the Das_t tables directly, while `ph5api.filter_das_t` filters/draws off the Das_t data read from the [ph5api.read_das_t ](https://github.com/PIC-IRIS/PH5/blob/master/ph5/core/ph5api.py#L874) method. The `ph5api.read_das_t` method internally calls the [experiment.ReceiversGroup. read_das](https://github.com/PIC-IRIS/PH5/blob/master/ph5/core/experiment.py#L696-L700) which programmatically changes the value of `sample_rate_multiplier_i`.

I'm still confused where the `sample_rate_multiplier_i` value is zero (0) in ph5toms. It seems like it should be programatically set to 1, but there must be another code path where it isn't. It seems like there are a few different code paths for reading the das table, making this difficult to debug.

hrotman-pic, 2021-01-27:
I have a question about the proposed utility in #446  for fixing the DAS tables: will it make a copy of the original DAS tables (in case of a mistake)?  I think a previous utility to update the DAS tables (fix_3chan_texan) created a copy of the original DAS table for all DAS as well as a copy of the updated DAS table.

timronan, 2021-01-27:
@hrotman-pic can you make this comment in the pull request? I don't think it is s bad idea to make a duplicate of the DAS table if a program is going to alter it. It is an additional safety check.

timronan, 2021-02-19:
This issue is resolved by merging PR #446 into the master branch.

closed 2021-02-19 by timronan