PIC-IRIS/PH5 #390: [BUG]

damhuonglan, 2020-03-27:
**Describe the bug**
metadatatoph5 creates response_t with response_file_das_a including info of both das model and sensor model while response_file_sensor_a has no info

**To Reproduce**
Run metadatatohp5 using ph5/test_data/metadata/station.xml
metadatatoph5 -n master.ph5 -f metadata/station.xml 
This will create response_t and array_t

Look in [response_t](https://github.com/PIC-IRIS/PH5/files/4395615/response.txt), with n_i=0:
 + response_file_das_a= /Experiment_g/Responses_g/NoneQ330_NoneCMG3T_200HHN
  + response_file_sensor_a=
With n_i=-1
  + response_file_das_a= 
  + response_file_sensor_a=

**Expected behavior**
The ‘None’ in the file name should be removed in [station.xml](https://github.com/PIC-IRIS/PH5/blob/master/ph5/test_data/metadata/station.xml) lines 223, 32,33
From [array_t](https://github.com/PIC-IRIS/PH5/files/4395616/array.txt), the response_file should be: 
With n_i=0:
  + response_file_das_a=/Experiment_g/Responses_g/Q330_200_1_1
  + response_file_sensor_a=/Experiment_g/Responses_g/CMG3T
With n=-1:
  + response_file_das_a=/Experiment_g/Responses_g/Q330_0_1
  + response_file_sensor_a=/Experiment_g/Responses_g/CMG3T

damhuonglan, 2020-03-27:
@akram-pic @ascire-pic: When should response n_i should be -1. I currently don't implement n_i=-1 in resp_load. I need this part to be clarified to put in resp_load.

ascire-pic, 2020-04-01:
@damhuonglan Honestly I don't know when response_n_i should be -1. I do not believe I have ever seen it as -1 in any of the PH5s I have built. It normally starts at 0 and increases from there. I have not used metadata to PH5 very much though.

nick-falco, 2020-04-01:
> @damhuonglan Honestly I don't know when response_n_i should be -1. I do not believe I have ever seen it as -1 in any of the PH5s I have built. It normally starts at 0 and increases from there. I have not used metadata to PH5 very much though.

@damhuonglan I'd have to look at the code, but I think you use -1 for LOG channels that don't have an instrument response.

damhuonglan, 2020-04-02:
@nick-falco Thanks Nick. I've just looked at the metadatatoph5.py code and see that:
+ With LOG channels, array's channel_number_i will be -2 (line 201). There is nothing with response's n_i here.
+ With channel that has no response (line 286) or with response_stages<=0  (line 281), response's n_i=-1. I don't know what is response_stages here. But anyway with response's n_i=-1, I don't have to care about response files any more, do I?
Another part I want to ask you is in resp_load.load_response(), line 471, array's response_table_n_i = None when true sample rate < 1. Why is that, should I keep it that way? I was about to give response_n_i for it in PR #373. If it is possible please clarify this part for me. Thank you.

damhuonglan, 2020-04-03:
@ascire-pic Can you have a look at station.xml please? It only have one response tag for each channel. Maybe it's the reason for having the response das and sensor combine together. In another hand, this is not a bug? In your experience, do you think a response entry with response_file_das_a= "/Experiment_g/Responses_g/Q330_CMG3T_200HHN" and response_file_sensor_a="" is acceptable?

damhuonglan, 2020-04-07:
After talking with Alissa, this is not a bug.

closed 2020-04-07 by damhuonglan