PIC-IRIS/PH5 #395: [BUG] seg2toph5, time increasing to finish each file

hrotman-pic, 2020-04-15:
**Describe the bug**
Using the command line for seg2toph5, the time needed to finish updating external references increases significantly with each file in a way that is noticeable after only a few files.  The most relevant parts of the seg2toph5.log:
[2020-04-15 17:48:10,801] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15001.dat...
[2020-04-15 17:48:15,705] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:48:16,762] - ph5.utilities.seg2toph5 - INFO: Done, 60 nodes recreated
[2020-04-15 17:48:16,763] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15002.dat...
[2020-04-15 17:48:20,705] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:48:24,691] - ph5.utilities.seg2toph5 - INFO: Done, 120 nodes recreated.
[2020-04-15 17:48:24,691] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15003.dat...
[2020-04-15 17:48:28,744] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:48:37,486] - ph5.utilities.seg2toph5 - INFO: Done, 180 nodes recreated.
[2020-04-15 17:48:37,486] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15004.dat...
[2020-04-15 17:48:41,599] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:48:56,901] - ph5.utilities.seg2toph5 - INFO: Done, 240 nodes recreated.
[2020-04-15 17:48:56,901] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15005.dat...
[2020-04-15 17:49:01,196] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:49:25,168] - ph5.utilities.seg2toph5 - INFO: Done, 300 nodes recreated.
[2020-04-15 17:49:25,168] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15006.dat...
[2020-04-15 17:49:29,503] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:50:03,518] - ph5.utilities.seg2toph5 - INFO: Done, 360 nodes recreated.
[2020-04-15 17:50:03,519] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15007.dat...
[2020-04-15 17:50:08,230] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:50:54,333] - ph5.utilities.seg2toph5 - INFO: Done, 420 nodes recreated.
[2020-04-15 17:50:54,334] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15008.dat...
[2020-04-15 17:50:58,920] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:51:59,246] - ph5.utilities.seg2toph5 - INFO: Done, 480 nodes recreated.
[2020-04-15 17:51:59,246] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15009.dat...
[2020-04-15 17:52:03,935] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:53:19,829] - ph5.utilities.seg2toph5 - INFO: Done, 540 nodes recreated.
[2020-04-15 17:53:19,829] - ph5.utilities.seg2toph5 - INFO: Processing: /ph51/Working/Holly_seg2_test/RAW/15010.dat...
[2020-04-15 17:53:24,569] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2020-04-15 17:54:57,885] - ph5.utilities.seg2toph5 - INFO: Done, 600 nodes recreated.
[seg2toph5_test_PIC.log](https://github.com/PIC-IRIS/PH5/files/4483696/seg2toph5_test_PIC.log)

**Environment (please complete the following information):**
 - OS: ph51
 - Program Version: seg2toph5 2019.42

**To Reproduce**
seg2toph5 -f ../seg2.list -n master.ph5 -M 5 -S 00001 where seg2.list includes the 10 unique files on the PIC Cloud under Public Uploads/Goldmand_ph5bug_seg2.  Each file has approximately 2 s of data from 60 DAS S/N on the PIC StrataVisor.

**Expected behavior**
Little or no increase in the time to update external references with additional files.  In other command-line routes to add data to a PH5, for example segdtoph5, the time to update external references increases very little over the same number of files.

SEG-2 data in this case is in a total of ~1500 files, but other data sets present and future can be expected to have more files.

dsentinel, 2020-04-15:
Thanks @hrotman-pic

dsentinel, 2020-04-20:
@hrotman-pic , did Mark mention a solution with obspy2ph5?

hrotman-pic, 2020-04-20:
The updating external references didn't slow and hang with obspy2ph5 like seg2toph5.

hrotman-pic, 2020-05-14:
Good news for comparison: I found the seg2toph5 log for TIME.  Some files at the start appear to have been added in separate seg2toph5 commands, unlike the original post example.

Some relevant parts:
[2019-02-20 17:38:21,871] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1007.dat...
[2019-02-20 17:38:31,091] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:38:31,627] - ph5.utilities.seg2toph5 - INFO: Done, 48 nodes recreated.
[2019-02-20 17:39:50,264] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1008.dat...
[2019-02-20 17:39:55,883] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:39:58,004] - ph5.utilities.seg2toph5 - INFO: Done, 96 nodes recreated.
[2019-02-20 17:40:53,913] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1009.dat...
[2019-02-20 17:40:59,670] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:41:04,545] - ph5.utilities.seg2toph5 - INFO: Done, 144 nodes recreated.
[2019-02-20 17:41:14,300] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/10010.dat...
[2019-02-20 17:41:14,310] - ph5.utilities.seg2toph5 - ERROR: . Can't process ../shots_seg2/10010.dat
[2019-02-20 17:41:30,513] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1010.dat...
[2019-02-20 17:41:36,447] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:41:44,745] - ph5.utilities.seg2toph5 - INFO: Done, 192 nodes recreated.
[2019-02-20 17:41:55,966] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1015.dat...
[2019-02-20 17:42:02,501] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:42:15,370] - ph5.utilities.seg2toph5 - INFO: Done, 240 nodes recreated.
[2019-02-20 17:43:28,792] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1016.dat...
[2019-02-20 17:43:35,686] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:43:54,771] - ph5.utilities.seg2toph5 - INFO: Done, 288 nodes recreated.
[2019-02-20 17:44:14,546] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1017.dat...
[2019-02-20 17:44:21,872] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:44:48,823] - ph5.utilities.seg2toph5 - INFO: Done, 336 nodes recreated.
[2019-02-20 17:44:48,823] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1018.dat...
[2019-02-20 17:44:56,293] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:45:30,495] - ph5.utilities.seg2toph5 - INFO: Done, 384 nodes recreated.
[2019-02-20 17:45:30,495] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1019.dat...
[2019-02-20 17:45:38,452] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:46:21,942] - ph5.utilities.seg2toph5 - INFO: Done, 432 nodes recreated.
[2019-02-20 17:46:21,942] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1020.dat...
[2019-02-20 17:46:30,061] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:47:22,620] - ph5.utilities.seg2toph5 - INFO: Done, 480 nodes recreated.
[2019-02-20 17:47:22,620] - ph5.utilities.seg2toph5 - INFO: Processing: ../shots_seg2/1021.dat...
[2019-02-20 17:47:31,082] - ph5.utilities.seg2toph5 - INFO: Updating external references...
[2019-02-20 17:48:36,629] - ph5.utilities.seg2toph5 - INFO: Done, 528 nodes recreated.

hrotman-pic, 2021-05-12:
Resolved by #440.

closed 2021-05-12 by hrotman-pic