PIC-IRIS/PH5 #398: [BUG] ph5 validate not giving error when no response

hrotman-pic, 2020-05-04:
**Describe the bug**
ph5_validate is no longer providing an error when a response is not referenced for a station-channel combination.

**Environment (please complete the following information):**
Possibly restricted to most recent ph5 updates at both PIC and DMC.

**To Reproduce**
Run ph5_validate on any archive that either does not have responses loaded or has response_table_n_i pointing at an index that does not exist in response.kef

**Expected behavior**
ph5_validate should give an error when a response is not found for a station-channel combination.  This problem appears to have arisen in the past few weeks, based on recent ph5_validate runs that failed to catch in the last week at PIC and DMC and a ph5_validate that @akram-pic mentioned did catch it in the last 2-4 weeks.

We can still verify whether or not a response is referenced by using ph5tostationxml at the response level.

damhuonglan, 2020-05-29:
Might be related to PR #383

hrotman-pic, 2020-06-01:
> Might be related to PR #383

I'm confused: this bug might be addressed in that PR?

damhuonglan, 2020-06-01:
In that PR, ph5_vadilate gives an error message if ph5 has no response. Please checkout branch i334_validate_err_resp to test it.

hrotman-pic, 2020-06-01:
Oh!  OK.  Thank you, that's great.  I think we can leave this issue on hold until that PR is merged, and after it is merged I can close this issue?

damhuonglan, 2020-06-01:
Yes, we only close the issue after we merge the branch to master. But please test so we know if it solves this issue or not. Thank you

damhuonglan, 2020-10-27:
solved in PR#383

hrotman-pic, 2020-12-23:
Re-opening this since it was resolved in the same PR as issue 334 and may be affected by any rollback done to resolve issue 443.

hrotman-pic, 2021-04-26:
Resolved by PR#445.

closed 2021-04-26 by hrotman-pic