PIC-IRIS/PH5 #399: [BUG] Worker parallelization is suspect; keyboard interupt only kills one worker not entire process

timronan, 2020-05-08:
**Describe the bug**
When running  `ph5tostationxml -p ./ -n master.ph5 --level response > all_stations.xml` a keyboard interrupt only kill one worker process and does not kill the main process that is spawning/controlling the worker. If a process is interrupted and a worker is killed how is it guaranteed that the worker's job is completed? Is this how the parallelization logic works across PH5?

**Environment (please complete the following information):**
 - CentOS Linux 7 (Core)
 - I am unsure how to find PH5 version, it is not printed in the programs help message. There is no --version flag. Most up to date.

**To Reproduce**
Run: 
`ph5tostationxml -p ./ -n master.ph5 --level response > all_stations.xml`
Do a keyboard interrupt. On a MAC OSX control C.

**Expected behavior**
The program should exit with error. Likely system.exit(1).  All pool workers should be canceled upon a keyboard interrupt. 

**Additional context**
Below is the output code from multiple keyboard interrupts. Note how only individual workers are killed. 
```
Process PoolWorker-110:
Process PoolWorker-112:
Traceback (most recent call last):
Traceback (most recent call last):
    self.run()
    self.run()
    self.run()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
    self._target(*self._args, **self._kwargs)
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
    self._target(*self._args, **self._kwargs)
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    racquire()
    racquire()
    racquire()
KeyboardInterrupt
    self.run()
KeyboardInterrupt
KeyboardInterrupt
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
    self.run()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    racquire()
    racquire()
KeyboardInterrupt
KeyboardInterrupt
^C**Process PoolWorker-118:**
**Process PoolWorker-114:**
Traceback (most recent call last):
Traceback (most recent call last):
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
    self.run()
    self.run()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
    self._target(*self._args, **self._kwargs)
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    racquire()
    racquire()
Process PoolWorker-117:
KeyboardInterrupt
KeyboardInterrupt
Traceback (most recent call last):
Process PoolWorker-116:
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
Traceback (most recent call last):
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
    self.run()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    self.run()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
    racquire()
    self._target(*self._args, **self._kwargs)
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
KeyboardInterrupt
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    racquire()
KeyboardInterrupt
Process PoolWorker-115:
Traceback (most recent call last):
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
    self.run()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/pool.py", line 102, in worker
    task = get()
  File "/home/PIC/.conda/envs/ph5/lib/python2.7/multiprocessing/queues.py", line 374, in get
    racquire()
KeyboardInterrupt
```

dsentinel, 2020-05-12:
This is unexpected behavior. I'm also curious why this would be optimized with parallel processes. Does extracting a StaXML take so long?

timronan, 2020-05-21:
I am not sure how long it takes to map between stationxml and kef. It seems to be dependent on the size of the metadata. Is the parallel logic embedded in more functions than ph5tostationxml? It doesn't seem like the parallel processing should be removed from ph5tostationxml, it just seems like the system that manages the parallel processes needs to be fixed to kill all of processes when one process is killed. It is possible that there an internal logging problem between the manager and drone and processes.

damhuonglan, 2020-10-12:
@timronan Does this still happen any more?

timronan, 2020-12-21:
This issue is none important and easily mitigated.

closed 2020-12-21 by timronan