PIC-IRIS/PH5 #401: [BUG] time_t does not reflect time/drift when same texan is used in multiple deployment

akram-pic, 2020-05-13:
**Describe the bug**
Time_t (/Experiment_g/Receivers_g/Time_t) is created by running 'time_kef_gen' to apply drift correction specifically for the rt125 (Texans). This table reports the time stamp, drift rate, and time correction by reading it from the raw TRD files (native texan). 

However, in case where multiple TRD files exist with the same serial number (from multiple deployments), PH5 fails to report time and drift for each file/texan, and only reports values for one. 

Example of this exist here: /ph52/Working/PLAYA1_201715/PH5/Sigma

For example in the experiment listed above, I have 6 TRD files for texan 2537, but the time_t only has values for "/I2537RAWLLAMA5R.TRD". 

* List of TRD files from the same texan in multiple deployments at different times:

> /ph52/Working/PLAYA1_201715/RAW/ALL-raw/S6-LabO23/I2537RAWSEISLAB1.TRD
> /ph52/Working/PLAYA1_201715/RAW/ALL-raw/Survey1/I2537RAWLLAMAN2.TRD
> /ph52/Working/PLAYA1_201715/RAW/ALL-raw/Survey2/I2537RAWLLAMAR3.TRD
> /ph52/Working/PLAYA1_201715/RAW/ALL-raw/Survey3/I2537RAWLLAMA4R.TRD
> /ph52/Working/PLAYA1_201715/RAW/ALL-raw/Survey4/I2537RAWLLAMA5R.TRD
> /ph52/Working/PLAYA1_201715/RAW/ALL-raw/Survey5/I2537RAW6RLLAMA.TRD

* The only entry in the "time_t" table for texan 2537 (which belongs to 'I2537RAWLLAMA5R.TRD' ):

>  Experiment_g/Receivers_g/Time_t
> 	das/serial_number_s = 12537
> 	start_time/epoch_l = 1493310643
> 	start_time/micro_seconds_i = 0
> 	start_time/ascii_s = Thu Apr 27 16:30:43 2017
> 	start_time/type_s = BOTH
> 	end_time/epoch_l = 1493757085
> 	end_time/micro_seconds_i = 0
> 	end_time/ascii_s = Tue May  2 20:31:25 2017
> 	end_time/type_s = BOTH
> 	slope_d = -2.63571e-08
> 	offset_d = -0.0117669

I can not comment at the moment on whether this is a new bug or has always existed. Requiers more digging in older archives.  

**To Reproduce**
See data and experiment at: /ph52/Working/PLAYA1_201715/PH5/Sigma 

**Expected behavior**
Time drift for each TRD file, at each deployment should be counted and reflected in the time_t table. 

**Additional context**
I understand this bug is very specific to the texans, their stand alone vcxo and how their time needs to be handled. I also realize they will phase out and will be replaced in the next couple of years. So priority in fixing this problem is something to discuss further.

damhuonglan, 2020-07-16:
@dsentinel In the last PH5 meeting, I asked DG to have some very small data that reflect this circumstances for unittest. However they said that they cannot create that but I can use the data that they encountered the problem at in the link above because they are quite small. I check that and see that I can use two raw data files, the size of each is 13.5MB. Do you think they are small enough for unittest?

dsentinel, 2020-07-16:
That's too big for unit tests. Can you chop the data down to a smaller size?

damhuonglan, 2020-07-16:
I don't know how to change raw data file. @akram-pic Do you have a way to chop the data down to a smaller size?

damhuonglan, 2020-07-17:
@akram-pic I found a way to add soh data to ph5 without using raw data. You don't have to worry about this.

akram-pic, 2020-07-31:
I have tested time-kef-gen and randomly selected few texans and looked at the raw file, its time drift during the operation in multiple arrays, and the time-t from the bug fix. Time-t now reflects all of the texans, and their time drift while they have been deployed in multiple arrays (same texan in separate deployment at different times). time-t is correct now. 

I have then extracted segy files (ph5toevt) as a single trace and gather with and without the -N (--notimecorrect) option. From the ph5toevt.log, I get a message that time drift correction is being applied accordingly for each trace. So this good. However, I do not have a good way of testing and confirming that the time drift is being applied correctly to the actual trace. 
Thank you.

PR #408 (2020-09-18): fix bug that time_t does not reflect time/drift when same texan is used in multiple deployment
### What does this PR do?
Fix #401 that time_t only has one entry for each das.

### Reason
Process_soh() looped through all soh for a das to get (to, from) and return the last (to, from) in the loop to be process to create one entry in time_t.

### Solution
Append all (to, from)s to a list to return. In main, each pair in the list will then be processed to create one entry in time_t.

### Checklist
- [ ] This PR is not directly related to an existing issue (which has no PR yet).
- [ ] All tests pass [... cut]
closed 2020-09-18 by damhuonglan