PIC-IRIS/PH5 #403: [BUG] list index out of range when there are repeated station in array table

damhuonglan, 2020-06-09:
**Describe the bug**
There was a reported bug on pforma that duplicate some stations in array_t (#360). It has been fixed in PR #361. However, there are still experiments with stations duplicated.

The old ph5_validate didn't have any problem with this issue. But with the current ph5_validate the duplicated stations will cause error "list index out of range".

**To Reproduce**
Experiment with duplicated stations can be found at 
/Desktop/ph51/completed/20-009_Playa2
or
/ph51/completed/18-028_BASIN

**Expected behavior**
ph5.utilities.ph5validate - ERROR: list index out of range

dsentinel, 2020-06-09:
If I understand, this is not a bug in the software, but some old PH5's are malformed?

timronan, 2020-06-11:
@damhuonglan the *malformed* PH5s stored at IRIS DMC may, and likely do, have errors. We set up a backdoor in our loader software to load historic  PH5 metadata regardless of its ability to pass the stationxml validator. We had to do this during the PH5 metadata migration because there are a significant portion of historical PH5 expirments that don't pass validation. Duplicated stations lead to 111 errors.

dsentinel, 2020-06-15:
The PR's have fixed the issue, but not for the data in the archive. I'm leaving this open.