PIC-IRIS/PH5 #409: [BUG] mstoph5 not correctly adding data to ph5 archive

ascire-pic, 2020-06-19:
mstoph5 was used to added miniseed format data to a PH5 archive. The data in question was the 2000sps SmartSolo shakedown data. The original miniseed data was 47Gb but the resulting ph5 archive was only 2Gb. mstoph5 appeared to run correctly based on the logs: 
[mstoph5-log.txt](https://github.com/PIC-IRIS/PH5/files/4806017/mstoph5-log.txt)

The DAS tables have the correct number of samples (~6 hours of recording at 2kHz sample rate = ~40,000,000 samples): 
[das-5500.txt](https://github.com/PIC-IRIS/PH5/files/4806025/das-5500.txt)

After extracting the data and plotting it, it appears that most of the samples have been flattened to zero. In attached plot, top trace is the original miniseed data. Bottom trace is miniseed extracted from the PH5 archive for the same station:
[image]

Bug was reproduced by running mstoph5 on the same data twice as well as isolating a single miniseed file and running mstoph5 on that single file. 

PH5 software on this computer was updated on 10 June 2020
 - OSX 10.10.5
 - mstoph5 Version: 2019.65

dsentinel, 2020-06-22:
My first guess would be that ms2ph5 can not handle floating point encoded ms.

ascire-pic, 2020-06-22:
Just ran mstoph5 on data at sps 1Hz and 100Hz from a Q330 and the data look correct when writing back out from the PH5. So if the SmartSolo data are encoded differently that might explain the problem.

PR #419 (2020-09-18): Fixing mstoph5 not correctly adding data to ph5
### What does this PR do?
Changing datatype according to raw datatype to fix #409 in which mstoph5 round float data to integer.

### Checklist
- [ ] This PR is not directly related to an existing issue (which has no PR yet).
- [ ] All tests pass.
- [ ] Any new or changed features have are documented.
- [ ] Changes have been added to `CHANGELOG.txt` .
- [ ] First time contributors have added your name to `CONTRIBUTORS.txt` .
closed 2020-09-18 by damhuonglan