PIC-IRIS/PH5 #413: [FEATURE]

jt38, 2020-07-07:
**Is your feature request related to a problem? Please describe.**
ph5toms provides options to filter exported data by networks, stations and SEED channels but not SEED locations.

**Describe the solution you'd like**
Allowing to filter data by location by providing a command line option to ph5toms.

dylanmikesell, 2020-07-24:
I would like to bump this feature request/enhancement. We have a seismo-acoustic experiment in Idaho that we are archiving as PH5. Location IDs are going to be heavily utilized in this archive because in addition to colocated seismic and acoustic sensors, the acoustic sensors are actually small arrays with location ID ranging from 00 to 20.

timronan, 2020-08-14:
I will look into this issue next week.

timronan, 2020-12-22:
Is this an important issue? Do ph5 datasets use location codes?

nick-falco, 2020-12-22:
The issue with PH5's lack of location code support is because sometimes PIs have have an experiment that includes broadband data with location codes as well as geophone data. Because PH5 doesn't support location codes, the broadband data has to be archived as miniSEED and given a different network code.

dylanmikesell, 2020-12-22:
So then is the recommendation to put nodal geophone data into the broadband seed archive if we have heterogeneous data like this? It seems to me that if the data were collected in the same region during the same campaign (and with the same sample rate), then they should have the same network code when that make sense.

timronan, 2020-12-22:
@dylanmikesell "data collected in the same region during the same campaign" should only be assigned one network code. Although experiments that contains both broad band and nodal data have historically been split between network codes, these splits have been reversed and these split experiments have been combined and assigned a consolidated network code. We have found that split data sets are difficult to manage and exchange. 

In theory, putting an experiment's nodal and broadband data into PH5 is the most desirable solution, but this does not work well because of issue #402. Because PH5 assumes that every channel/location can only be assigned a parent station if both the channel/location and the station have the same coordinates, it is not possible to describe a data logger connected to multiple non-colocated instruments. This issue make it impossible to accurately describe a number of different geophysical methodologies (E.g. Infrasound, DAS) using PH5. 

It seems like including location code capabilities into PH5 may not be  worth the effort unless nodal data somehow uses location codes or issue #402 can be resolved, the latter does not seems possible unless the PH5 data structure is altered. I am wondering if there is any nodal specific reasons to include location code capabilities into PH5?

closed 2022-11-15 by timronan