PIC-IRIS/PH5 #432: [BUG] ph5tostationxml outputs stationxml that is does not comply with the stationxml xsd schema

timronan, 2020-10-08:
**Describe the bug**
Ph5tostationxml outputs stationxml that is does not comply with the stationxml xsd schema. 

Specifically the field :
<Created>2020-10-08 21:58:09.039103</Created>

Is formatted incorrectly. 

It should at least have a T inbetween the day and hour, similar to:
<Created>2020-10-08T21:58:09.039103</Created>

This bug can be quickly addressed by changing the format
created=datetime.now() on line 955 clients.ph5tostationxml
to
created=datetime.now().isoformat(sep='T')
or 
created=datetime.now().strftime('%Y-%m-%dT%H:%M:%S')

**Environment (please complete the following information):**
 - OS: Centos 7
 - Program Version oph5tostationxml  Version: 2019.63

**To Reproduce**
Steps to reproduce the behavior:
run ph5tostationxml 
run the output stationxml document through the validator. 
**Expected behavior**
Output stationxml doucment should pass validation. 

**Additional context**
Issue started to occur after updating obspy. We should be explicit with our <Created> date times so we don't run into this issue.

nick-falco, 2020-10-09:
@timronan I think the correct date string format should be:

```
'%Y-%m-%dT%H:%M:%S.%f'
```

Note the `.%f` at the end of the date string.

```
>>> from datetime import datetime
>>> created = datetime.now().strftime('%Y-%m-%dT%H:%M:%S.%f')
>>> print (created)
2020-10-09T11:47:25.525567
```

timronan, 2020-10-09:
@nick-falco thanks for the catch.

nick-falco, 2020-10-09:
Actually it may be better to use the Obspy UTCDateTime class instead and also put a "Z" at the end of the format string. I think the times should be in UTC.

ascire-pic, 2020-10-13:
@timronan Are you planning to work on this issue?

timronan, 2020-10-13:
@ascire-pic I can fix this issue but I need to finish all of my outstanding projects before I start on this one. There may be a delay before I can work on this patch.

ascire-pic, 2020-10-13:
Sounds good! Was looking through the open issues and just wanted to check.

gbbofh, 2020-10-23:
@timronan  I can start on a pull request to fix this if you haven't started working on it yet.

timronan, 2020-10-23:
@gbbofh please start this PR. Thew fix seems to be outlined in this issue.

gbbofh, 2020-10-26:
@timronan I should have a PR fixing this issue up later today.

timronan, 2020-10-27:
@gbbofh sounds good. I will review the PR when it is posted.

closed 2021-02-23 by timronan