PIC-IRIS/PH5 #433: Modifications to ph5_validate checks

hrotman-pic, 2020-10-13:
This issue is to request changes to things that ph5_validate checks.  I will organize by category.  This is from the list that @damhuonglan researched.

Experiment_t: 
Nickname not found is currently a warning.  Request change to error since the nickname is used to name SEGY files written with ph5toevt.

Array_t: 
'Station ID not between 0 and 65535' and 'Station ID not a whole number between 0 and 65535': request that only one of these is used, whichever is more specific and informative.
Request warning if station ID is less than 65535 and more than 32767 to accommodate segy1 compatibility.  Suggested warning text: 'Station  ID is more than 32767.  Not compatible with SEGY revision 1.'
Sample rate <=0 is currently a warning.  Request change: I agree with the idea of a negative sample rate being an error and a sample rate of 0 being a warning.  However, if sample rate=0 means data cannot be extracted (I think it does?), both cases should probably be an error.
Question on the sample rate multiplier check: should it flag for multiplier <=0 or <=1?
I suggest that compliance with SEED (band codes, instrument codes, orientation) conventions and their stationXML schema equivalents be deferred for later consideration in a subsequent iteration of PH5.

Event_t:
'Event_t table not found.  Did this experiment have shots???' is correctly a warning in stdout.  Request change so this is also a warning in the ph5_validate log file.
Requesting addition of check that event ID is a whole number between 0 and 65535, and error when it is not.
Because some users are still utilizing segy1, request warning if event ID is less than 65535 but more than the segy1 maximum of 32767.  Suggested warning text: 'Event ID is more than 32767.  Not compatible with SEGY revision 1.'
Remove the checks for 'No Event location/coordinate_system_s value found', 'No Event location/projection_s value found', 'No Event location/ellipsoid_s value found', and 'No Event location/description_s value found' because those fields do not exist in the event table. (See DG PH5 documentation and/or noven dropdown options in shot configuration.)

Both Array_t and Event_t: Request that the flagging for 'No location/Z/unit_s value found' be consistent.  It is currently listed as a warning in Array_t and error in Event_t.

Response:
If negative indices are not allowed in the response table, or not preferred, request error or warning flag.

Please discuss any other changes in comments.

damhuonglan, 2020-10-19:
> 'Station ID not between 0 and 65535' and 'Station ID not a whole number between 0 and 65535': request that only one of these is used, whichever is more specific and informative.
Only one of these will be reported. The first one is checking for the number in between 0 and 65535, error may occur when station id contains any letters or special characters, then the later will be reported. I don't think we should try to combine them.

hrotman-pic, 2020-10-20:
@ascire-pic and @akram-pic, would you please take a look at the list of changes and see if you have any suggestions or clarifications?

damhuonglan, 2020-10-21:
> 'Event_t table not found. Did this experiment have shots???' is correctly a warning in stdout. Request change so this is also a warning in the ph5_validate log file.

This is currently an an error in ph5_validate log file. I will change it to warning for consistency.

hrotman-pic, 2021-05-25:
Resolved by #435 .  Thank you again.

closed 2021-05-25 by hrotman-pic