PIC-IRIS/PH5 #450: [BUG] ph5toevt does not work correctly for SmartSolo data

hrotman-pic, 2021-03-10:
**Describe the bug**
ph5toevt does not populate headers and cannot create a readable SEGY gather for data ingested with mstoph5 from SmartSolo miniseed files.

**Environment (please complete the following information):**
 - OS:server ph52
 - Program Version ph5toevt 2019.66

**To Reproduce**
ph5toevt -n master.ph5 --starttime=2020:313:16:12:37.000 -A 001 -l 3 -o gather3
[2021-03-10 16:53:30,868] - ph5.clients.ph5toevt - INFO: 2019.66: ['/home/ph5/anaconda3/envs/ph5/bin/ph5toevt', '-n', 'master.ph5', '--starttime=2020:313:16:12:37.000', '-A', '001', '-l', '3', '-o', 'gather3']
[2021-03-10 16:53:30,882] - ph5.clients.ph5toevt - INFO: Experiment: CO2 containment and monitoring at Crystal Geyser
[2021-03-10 16:53:30,882] - ph5.clients.ph5toevt - INFO: Summary: Seismic imaging to characterize fault properties and outgassing signature related to a natural CO2 seep. Site is located near Green River, Utah.
[2021-03-10 16:53:30,882] - ph5.clients.ph5toevt - INFO: Extracting receivers for event 2020-11-08 16:12:37+00:00.
[2021-03-10 16:53:30,883] - ph5.clients.ph5toevt - WARNING: Warning: No shot to receiver distances found.
[2021-03-10 16:53:30,883] - ph5.clients.ph5toevt - INFO: -=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
[2021-03-10 16:53:30,883] - ph5.clients.ph5toevt - INFO: Attempting to find data for station 8066.
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO: Applied time drift correction by                        shifting trace by -0.0 samples.
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO: Correction is 0 ms.
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO: Clock drift (seconds/second): 0.0
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO: Clock comment: No clock drift information available.
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO: =-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO: trace: 1
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO: Extracted: Station ID 8066
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO: Chan: 1 Start: 2020-11-08 16:12:37+00:00, Stop: 2020-11-08 16:12:40+00:00.
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO: Lat: 38.939038 Lon: -110.136246 Elev: 1234.821000 m
[2021-03-10 16:53:30,905] - ph5.clients.ph5toevt - INFO:
[2021-03-10 16:53:30,907] - ph5.clients.ph5toevt - INFO: Opened: gather3/COTEC_001_2020-11-08 16:12:37+00:00_123_0001.SGY
[2021-03-10 16:53:30,907] - ph5.core.segyfactory - ERROR: Failed to set data array. local variable 'x_f' referenced before assignment
Traceback (most recent call last):
  File "/home/ph5/anaconda3/envs/ph5/bin/ph5toevt", line 11, in <module>
    load_entry_point('ph5', 'console_scripts', 'ph5toevt')()
  File "/home/ph5/PH5/ph5/clients/ph5toevt.py", line 655, in main
    gather(args, p5)
  File "/home/ph5/PH5/ph5/clients/ph5toevt.py", line 576, in gather
    trace, fh, sf, num_traces)
  File "/home/ph5/PH5/ph5/core/segyfactory.py", line 1134, in write_segy_hdr
    e.message))
ph5.core.segyfactory.SEGYError: (("Error: Failed to write reel and first trace. local variable 'nparray' referenced before assignment\n",), {})
Closing remaining open files:./master.ph5...done./miniPH5_00024.ph5...done

This command works on rt125 and Fairfield archives on the same server, with start time adjusted appropriately.

**Additional context**
This affects recent and in-progress archives 21-001 and 21-002.
Das tables list a stream_number_i=0 for archive 21-002.  For the Fairfield archive I checked, stream_number_i=1 in the das tables.  rt125 archives may also have stream_number_i=1 in the das tables.  I cannot immediately identify any other major differences in the das or array tables.

hrotman-pic, 2021-03-10:
stream_number_i=0 in the das tables for data ingested with mstoph5 where the miniseed was created with rt2ms.  I will attempt to clean up the associated archive (used for testing) and explore this some more.

hrotman-pic, 2021-03-12:
Modifying stream_number_i to 1 in the das tables (like das tables with a different ingestion method where the same ph5toevt option works), and modifying event_number_i to be nonzero and sequential (again, like other das tables) by themselves does not cause ph5toevt to work or change the error.

The solution may lie in a different direction.

timronan, 2021-03-12:
What experiment was this test run on? Which data are you referring too? This may be a problem in the data and not the software.

hrotman-pic, 2021-03-12:
I changed the das tables for a subset of data (2 DAS for 8 days) also loaded in PH5 21-002, and for a subset of SEED X6.2019 (both DAS, roughly 20 days).  The latter has been successfully archived.  For the former, I cannot see any problems with the native data.  The miniseed files ingested were created with two different methods: 21-002 is a data offload format for the SmartSolos, and X6.2019 was converted from the native RefTek format to miniseed.  @ascire-pic is familiar with PH5 21-001.

timronan, 2021-03-12:
How do I reproduce the stack trace posted above? Was that generated using 21-002 or 21-002 or something different?

hrotman-pic, 2021-03-12:
The one I posted is from 21-002.  I believe the error is consistent across archives created with mstoph5 (which also requires using metadatatoph5).

timronan, 2021-03-12:
Thanks, for expirment 21-002 the data fails ph5tostationxml. 

ph5tostationxml -p ./ -n master.ph5 --level response 
[2021-03-12 21:34:27,218] - ph5.clients.ph5tostationxml - INFO: CHECKING ./master.ph5
[2021-03-12 21:34:28,025] - ph5.clients.ph5tostationxml - INFO: STATIONXML DATA CREATED FOR ./master.ph5
[2021-03-12 21:34:28,032] - ph5.clients.ph5tostationxml - ERROR: 'Frequency' object has no attribute 'measurement_method'

timronan, 2021-03-15:
This issue does not seem to be urgent or really a bug... it may be an enhancement. PH5toEVT seems to work on the data that has been accepted into the IRIS archive thus far.

ascire-pic, 2021-10-05:
Closing this issue since we decided not to use mstoph5 to ingest SmartSolo data and instead modified segdtoph5 (see PR 451) to ingest data from the SmartSolos. As mstoph5 is on the list to remove from the code stack, closing this issue seems appropriate.

closed 2021-10-05 by ascire-pic