PIC-IRIS/PH5 #453: seg2toph5 not working after server updates

hrotman-pic, 2021-03-31:
**Describe the bug**
Command-line and GUI seg2toph5 ingestion attempts result in an error like the following:
`[2021-03-31 14:29:53,702] - ph5.utilities.seg2toph5 - ERROR: list indices must be integers, not unicode. Can't process /external/ph5/Working/Holly_seg2_test/RAW/15001.dat`
I also got the same error with a few SEG-2 files that were used in the 18-030 PH5.

**Environment (please complete the following information):**
 - OS: ph51
 - Program Version seg2toph5 2019.42

**To Reproduce**
Run pforma with a list of one or more SEG-2 (.dat) files, or initialize a ph5 and run `seg2toph5 -f list-of-files -n master.ph5 -M num_minis -S starting_mini`

**Expected behavior**
Expect ingestion of SEG-2 format files to occur.

**Additional context**
Recent updates included moving to obspy 1.2.2.
The ingestion works on a server that has not received these updates.
Other ingestion methods work (do not generate an error) on the updated server.

The files I was using do not need to go into a PH5 archive.

hrotman-pic, 2021-05-12:
@damhuonglan it looks like all the relevant PRs have been merged and this is solved.  Thank you again.  Should I close this?

damhuonglan, 2021-05-12:
@hrotman-pic Hold on. Maeva is updating PH52 server with Obspy 1.2.2 and PySide2. After that we will need you check PH52 once more to make sure everything ok. Then we can close this.

hrotman-pic, 2021-05-12:
@damhuonglan OK, I will wait.

hrotman-pic, 2021-05-25:
seg2toph5 works in command line and pforma on ph52 after updates as well.

PR #454 (2021-04-07): Make seg2toph5 compatible with both Obspy 1.1.1 and 1.2.2
### What does this PR do?
Fix #453 seg2toph5 not working after update conda environment to use Obspy 1.2.2

### Checklist
- [ ] This PR is not directly related to an existing issue (which has no PR yet).
- [ ] All tests pass.
- [ ] Any new or changed features have are documented.
- [ ] Changes have been added to `CHANGELOG.txt` .
- [ ] First time contributors have added your name to `CONTRIBUTORS.txt` .
closed 2021-05-25 by hrotman-pic