PIC-IRIS/PH5 #458: [BUG]Error when running create_ph5.sh

damhuonglan, 2021-05-12:
**Describe the bug**
When reviewing one check failed on Travis_CI, I saw some errors happening when running create_ph5.sh. It wasn't the cause of the failed test, but it should be fixed:
```
Traceback (most recent call last):
  File "/home/travis/miniconda/envs/ph5/bin/metadatatoph5", line 11, in <module>
    load_entry_point('ph5', 'console_scripts', 'metadatatoph5')()
  File "/home/travis/build/PIC-IRIS/PH5/ph5/utilities/metadatatoph5.py", line 513, in main
    f = open(args.infile, "r")
IOError: [Errno 2] No such file or directory: '../../response_table_n_i_dup/station_response.xml'

Traceback (most recent call last):
  File "/home/travis/miniconda/envs/ph5/bin/mstoph5", line 11, in <module>
    load_entry_point('ph5', 'console_scripts', 'mstoph5')()
  File "/home/travis/build/PIC-IRIS/PH5/ph5/utilities/obspytoph5.py", line 591, in main
    found_files = getListOfFiles(args.indir)
  File "/home/travis/build/PIC-IRIS/PH5/ph5/utilities/obspytoph5.py", line 445, in getListOfFiles
    listOfFile = os.listdir(dirName)
OSError: [Errno 2] No such file or directory: '../../response_table_n_i_dup/miniseed/'
```

damhuonglan, 2021-05-12:
@timronan You added this part in PR #445. I think you forgot to add folder `response_table_n_i_dup/` to `ph5/test_data/`.
It seems that the data created in `ph5/test_data/ph5/response_table_n_i_dup` are not used in unittest because I don't see unittest failed.
So can you please either add the folder `response_tabel_n_i_dup` or removed code from Lines 66 to 82? Thank you.
https://github.com/PIC-IRIS/PH5/blob/87a31a09ab217d2405ba6fff566324a5ae772e48/create_ph5.sh#L66-L82

timronan, 2021-05-14:
@damhuonglan I just opened PR #459 and removed the code block as suggested. Let's verify that Travis CI passes. If it does this seems ready to be reviewed and merged. I know making a branch to solve this issue seems kind of heavy, but I didn't want to push directly into master.

damhuonglan, 2021-05-18:
@timronan It passed Travis CI so I approved and merged it. Thank you.

closed 2021-05-18 by damhuonglan

timronan, 2021-05-18:
Fixed by PR #459