PIC-IRIS/PH5 #469: [BUG] ph5toms not writing some sac headers correctly

hrotman-pic, 2021-07-01:
**Describe the bug**
ph5toms is writing out SAC header CMPINC with SEED convention instead of SAC convention.
Examples from 20-001 extracted at PIC after PI inquired:
_DP1_
     CMPINC = 0.000000e+00
_DP2_
     CMPINC = 0.000000e+00
_DPZ_
     CMPINC = 9.000000e+01

In SAC, CMPINC is the "component incident angle (degrees from upward vertical)", which is different from the SEED convention of using dip which is degrees from horizontal down.

CMPINC values are the same on other PH5s sampled.

**Environment (please complete the following information):**
 - OS: PH5 server at PIC, data query to IRIS
 - Program Version ph5toms 2019.93

**To Reproduce**
This is probably reproducible on any PH5.

**Expected behavior**
CMPINC=90 for both horizontal components (such as DP1 and DP2), and CMPINC=180 for the vertical component for most sensors used in PH5 experiments.  (Sensors from a SEED experiment with the same conventions have headers like this in SAC from FDSNWS-dataselect.)

**Additional context**
This is a bug in the sense that output is inaccurate, but ph5toms is likely looking up the receiver table correctly--that is, the program is working as written.

timronan, 2021-07-06:
@hrotman-pic can you please provide the exact command and experiment to reproduce the issue?

hrotman-pic, 2021-07-09:
Oops, sorry about that.  A few examples from different experiments.

20-001 (built late April 2021):
ph5toms -n master.ph5 -a 001,009 -s 2020-10-27T01:00:00 -t 2020-10-27T01:15:00 -o sac_test -F SAC

19-029 (built mid Dec. 2019):
ph5toms -n master.ph5 -a 001 -s 2019-12-02T05:00:00 -t 2019-12-02T05:15:00 -o sac_test -F SAC

19-015 (built late Sept. 2019):
ph5toms -n master.ph5 -a 001 -s 2019-07-16T05:00:00 -t 2019-07-16T05:15:00 -o sac_test -F SAC

These queries all output data from all 3 channels.  I can try it with a list of SEED channels (--channel) or channel numbers (-c) to extract, if you think that would help.  And/or, reproduce these queries on web services for some unrestricted PH5 datasets.  I might also have some SAC files from older PH5WS requests and I could look at those headers as well.

For 20-001, do you want to know what command the PIs used and if it was web services or another tool?

closed 2023-05-02 by hrotman-pic