PIC-IRIS/PH5 #493: [BUG] SmartSolo array table coordinate precision decreased

hrotman-pic, 2022-01-05:
**Describe the bug**
The latitude and longitude in degrees are rounded to 3 decimal places on the SmartSolo array tables.  Previously (in August 2021) there did not appear to be any rounding of coordinates.  I checked the original SEG-D headers and they do not appear rounded.  For example, it looks like SEG-D headers of:
IGU_GPS_lat_integer 34
IGU_GPS_lat_fraction 0.0918731689453
IGU_GPS_lon_integer -106
IGU_GPS_lon_fraction 0.871368408203
Are in the array table as:
location/X/value_d=-106.871
location/Y/value_d=34.092

**Environment (please complete the following information):**
 - OS: PIC PH5 servers
 - Program Version segdtoph5 2021.159/pforma 2021.84

**To Reproduce**
Ingesting any SmartSolo data with up to date PH5 should reproduce.  The dataset I noticed this on is large (multiple TB), but a subset should work.

**Expected behavior**
No rounding (I think Fairfield coordinates are not rounded?); or, rounding to a greater precision.  For example, 6 decimal places would match the SEED standard which is enforced in the stationXML writeout, and it's moderately common for me to see PI-provided coordinates that are surveyed to 7 decimal places.

**Additional context**
This is after PR484 (update for signed header) and PR479 (combined array entries) were merged, but I am not sure which one introduced the rounding.

derick-hess, 2022-01-10:
Looks like it's a simple bug in ph5/core/segdreader_smartsolo.py

lines 160-163 specifically format it to 3 decimal places.

hrotman-pic, 2022-02-08:
Resolved by #495 .

closed 2022-02-08 by hrotman-pic