PIC-IRIS/PH5 #497: [BUG] ph5toevt & ph5torec not handling overlapping data correctly

ascire-pic, 2022-01-25:
ph5toevt (and ph5torec) are not handling overlaps in data in a way that results in usable SEGY files. If an overlap exists in the data, ph5toevt writes the extra samples into the SEGY trace which creates invalid SEGY because the SEGY trace has more samples than are indicated in the trace header. This behavior and writing of extra samples is actually noted in the ph5toevt output:

[2022-01-19 23:06:25,743] - ph5.clients.ph5toevt - INFO: =-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-
[2022-01-19 23:06:25,743] - ph5.clients.ph5toevt - INFO: trace: 12
[2022-01-19 23:06:25,743] - ph5.clients.ph5toevt - INFO: Extracted: Station ID 12
[2022-01-19 23:06:25,743] - ph5.clients.ph5toevt - INFO: Chan: 1 Start: 2020-08-09 01:09:55+00:00, Stop: 2020-08-09 01:10:05+00:00.
[2022-01-19 23:06:25,744] - ph5.clients.ph5toevt - INFO: Lat: 30.169000 Lon: -90.801000 Elev: 0.000000 unknown
[2022-01-19 23:06:25,744] - ph5.clients.ph5toevt - INFO: DAS: 1X12, Node ID: 453005801
[2022-01-19 23:06:25,744] - ph5.clients.ph5toevt - INFO: Wrote: 5001 samples with -1 sample padding.
[2022-01-19 23:06:25,744] - ph5.clients.ph5toevt - INFO: Only wrote 5001 samples.
[2022-01-19 23:06:25,745] - ph5.clients.ph5toevt - INFO: Done.

In the example above, the ph5toevt output states that it is writing 5001 samples (the -1 sample padding, ie a 1 sample overlap), but the SEGY trace header indicates there are only 5000 samples in that trace. This causes issues with reading/plotting the SEGY as the number of samples in the trace header is used to specify when a SEGY reader should expect to find the next headers. As a result, any SEGY headers/data after a trace with an extra sample (or multiple extra samples) are interpreted incorrectly and the file becomes unreadable.

Note that all testing of this issue has been done with ph5toevt, but it is assumed a similar problem exists in ph5torec as they were written at the same time.

ph5toevt version 2019.66
ph5torec version 2019.66

Data set referenced above is on the ph5 server:
/external/ph5/Working/test-alissa/Sorrento_SegD/pforma-archive/Sigma

ph5toevt command with overlapping data on the above data set:
>> ph5toevt -n master.ph5 -A 1 -s 2020:222:01:09:55 -l 10 --use_deploy_pickup -c 1 -o gathers/
The overlapping sample is at 2020:222:01:10:00

Desired behavior: 
When ph5toevt/ph5torec encounters overlapping data, the extra samples should be discarded and the user informed. That will allow the user to still receive a valid SEGY file.

timronan, 2022-01-28:
Is this issues relates to issue #485

ascire-pic, 2022-01-31:
Not exactly related to #485? Technically if you wrote out SEG-Y for the time period specified in the example in 485 it would be invalid because of this issue but that would be true for any data with an overlap, not just SmartSolo data discussed in 485.

closed 2022-05-31 by ascire-pic