PIC-IRIS/PH5 #507: [BUG] segytoph5 throwing exceptions an not running

mike-iris, 2022-10-05:
**Describe the bug**
segytoph5 throwing exceptions an not running

**Environment (please complete the following information):**
 MacOS 12.6

**To Reproduce**
run in the regular git deployed configuration

**Expected behavior**
read a segy file and create a respective set ph5, master and miniPH5 files without metadata

**Screenshots**
n/a

**Additional context**
n/a

closed 2022-11-15 by timronan

timronan, 2022-11-15:
`segytoph5` is not currently being supported. The transformation between SEGY and a modern active source storage format maybe considered at a future date.

timronan, 2022-11-15:
This code could be potentially removed from the PH5 software package. Refer to issue `https://github.com/PIC-IRIS/PH5/issues/348`.