PIC-IRIS/PH5 #511: [BUG] ph5availabilty potential duplicating availabilty information

timronan, 2022-12-07:
**Describe the bug**
ph5availability may be reporting inaccurate duplicate results for data traces. We are unsure if PH5 availability is accurately outputting information that is stored in the ph5 experiment or if, a duplicate availability record is being produce. ph5availabilty and ph5toms are producing inconsistent results in regards to data duplication. 

Can someone check to see if ph5avaibility is accurately reflecting the data or if the duplicate record is somehow being created?

**Environment (please complete the following information):**
ph5availability commit 33f9d6d2621fb5550f23d9a8452ef18a9a744880 (most up date commit)

**To Reproduce**
Steps to reproduce the behavior:
**On experiment 19-024**
```
ph5availability -p . -n master.ph5 --station 1113 --channel GP1 -s 2019-08-05T22:17:00 -e 2019-08-08T16:45:00 -a 2
[2022-12-07 05:11:42,566] - ph5.clients.ph5availability - WARNING: Using sample rate from DAS Table 1X1113.
[2022-12-07 05:11:42,566] - ph5.clients.ph5availability - WARNING: Using sample rate from DAS Table 1X1113.
[2022-12-07 05:11:42,566] - ph5.clients.ph5availability - WARNING: Using sample rate from DAS Table 1X1113.
[2022-12-07 05:11:42,567] - ph5.clients.ph5availability - WARNING: Using sample rate from DAS Table 1X1113.
skipping exception and running query DAS T
#n s     l  c   q                    earliest                      latest
1B 1113  -- GP1   2019-08-05T22:17:00.000000Z 2019-08-08T16:45:00.000000Z
1B 1113  -- GP1   2019-08-05T22:17:00.000999Z 2019-08-08T16:45:00.000000Z
```

**Expected behavior**
For ph5availability to match what is contained in the experiments

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
This problem is not associated with the optimization done in PR #509

hrotman-pic, 2022-12-12:
I've checked the array and DAS tables for this experiment and the array entries for this station are not duplicated and the loaded data are not duplicated.  I confirmed no duplication in the miniseed output.

timronan, 2023-01-10:
This may issue #499. Retest this issue after we merge PR #505.

closed 2023-05-02 by timronan