PIC-IRIS/PH5 #523: [BUG] ph5toevt: Event table entries with a numeric description_s case ph5toevt crash with attempting to write files with

Stephen-Veitch, 2023-12-26:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment (please complete the following information):**
 - OS: [e.g. Ubuntu 16.04]
 - Program Version [ph5toms 2018.268]

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Stephen-Veitch, 2023-12-26:
[ph5toevt_error.txt](https://github.com/PIC-IRIS/PH5/files/13772952/ph5toevt_error.txt)

damhuonglan, 2024-01-23:
@Stephen-Veitch I modified the code so it print out the error. This is the new error file: 
[new_ph5toevt_error.txt](https://github.com/PIC-IRIS/PH5/files/14028617/new_ph5toevt_error.txt)

The error line is:
```
[2024-01-23 18:44:55,653] - ph5.core.segyfactory - ERROR: Failed to write extended portion of SEG-Y trace header: 'H' format requires 0 <= number <= 65535
```
It still isn't clear but it reports the number in `event_t['description_s']` to be `0 <= number <= 65535` while it is currently `14530055261`.
Currently the structure for this field in SEGY is INT16. If I change to INT32, no error will happen. But I don't have the format for SEGY extended trace header style Menlo USGS to check if that is correct or not. But I believe it is.

I have a suggestion for this which is to check the description, if it satisfies the limit range it will be added to segy, if not it won't be added to segy and a warning will be added as following
```
[2024-01-23 19:01:35,892] - ph5.core.segyfactory - WARNING: event_t[description_s]=453005526 is ignored because it doesn't satisfy the limit range [0, 65535] 
```
How do you think about the suggestion above?

Stephen-Veitch, 2024-01-23:
Suggest moving forward with this fix with warning message as follows:
"WARNING: description_s not added to segy header: Descriptions must be numeric values in range [0,65535] to be added to header."

closed 2024-03-12 by damhuonglan