PIC-IRIS/PH5 #550: queries for start time 00:00:00 result in miniseed that starts at 23:59:59 the previous day

gillian-earthscope, 2025-01-17:
Back in 2019, Nick was working on a fix to ensure that ph5 data coming out of dataselect did not include samples prior to the requested start time. It's not always working as expected now: if I request data for 00:00:00 it can return data starting 23:59:59.999999 the day before for some (but not all) stations. Did something change or did Nick's solution not get fully implemented? I start seeing this commonly since August 2022 in MUSTANG QC metrics, but there are a few instances on 2020-02-11. It is causing problems in the MUSTANG system that we'll have to work around.

Example:
http://service.iris.edu/ph5wsbeta/dataselect/1/query?reqtype=FDSN&format=mseed&net=YW&sta=5002&loc=--&cha=DPZ&starttime=2016-07-11&endtime=2016-07-12&nodata=404

trace download from Nick's fix:
=> msi -tg /Users/nick/Downloads/ph5ws-dataselect_2019-06-07t18_59_32z.mseed
  Source         Start sample        End sample     Gap  Hz  Samples
YW_5002__DPZ    2016,193,00:00:00.000000 2016,193,23:59:59.995998  ==  250 21600000

same query yesterday:
=> msi -tg ph5ws-dataselect_2025-01-17t04_44_22z.mseed
   Source                Start sample             End sample        Gap  Hz  Samples
YW_5002__DPZ      2016,192,23:59:59.999999 2016,193,23:59:59.995999  ==  250 21600000

hrotman-pic, 2025-01-21:
Hi Gillian, are there some other example experiments & stations you're seeing the start of 23:59:59.999999 on with dataselect?  Depending on the experiments I can try to reproduce at EPIC, and I can work on identifying if there's any commonalities with the experiments that have this issue.

hrotman-pic, 2025-01-21:
We did some testing during the PH5 meeting.  For 20-003, extracting a day of DPZ for all stations

ph5toms -n master.ph5 -c 3 -s 2020-01-17T00:00:00 -t 2020-01-18T00:00:00 -o mseed_test2_allc3 -f MSEED

On the resulting mseed files:
msi -S XH*
DCC|2025,021
XH|1101||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1102||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1103||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1104||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021
XH|1105||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021
XH|1106||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995999||250|21600000|||||||2025,021
XH|1107||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995999||250|21600000|||||||2025,021
XH|1108||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1109||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021
XH|1110||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995999||250|21600000|||||||2025,021
XH|1111||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1112||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1113||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1114||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1115||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995999||250|21600000|||||||2025,021
XH|1116||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1117||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995999||250|21600000|||||||2025,021
XH|1118||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021
XH|1119||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1120||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1121||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021
XH|1122||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021
XH|1123||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1124||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995999||250|21600000|||||||2025,021
XH|1125||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021
XH|1126||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1127||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1128||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021
XH|1129||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021
XH|1130||DPZ|2020,017,00:00:00.000000|2020,017,23:59:59.996000||250|21600000|||||||2025,021
XH|1131||DPZ|2020,016,23:59:59.999999|2020,017,23:59:59.995998||250|21600000|||||||2025,021

timronan, 2025-01-21:
This bug may be associated with this method:

https://github.com/PIC-IRIS/PH5/blob/master/ph5/core/ph5api.py#L1107

gillian-earthscope, 2025-01-21:
[1H.list.txt](https://github.com/user-attachments/files/18496350/1H.list.txt)
I tried uploading a file with all PH5 that were processed by Mustang and had start times recorded as either 00:00:00 or 23:59:59 of the previous day, but it was too large. Here is network 1H: the file is target, data start time, Mustang processing time (lddate). This data was always requested as 00:00:00 to 23:59:59.999999 of a UTC day.

The file shows mixed results for targets and for same-day Mustang runs. One target that I explored further is two days of 1H.3002..GP1.D (same run, different behavior):
"target", "start", "lddate"
"1H.3002..GP1.D","2021/12/15 23:59:59","2022/11/06 13:22:30"
"1H.3002..GP1.D","2021/12/22 00:00:00","2022/11/06 13:12:29"

I requested the data from ph5ws/dataselect and this time both days returned start times of 23:59:59. 
> wget -O mseed1 'https://service.iris.edu/ph5ws/dataselect/1/query?reqtype=FDSN&format=mseed&net=1H&sta=3002&cha=GP1&starttime=2021-12-16&endtime=2021-12-17&nodata=404'
> msi -tg mseed1
   Source                Start sample             End sample        Gap  Hz  Samples
1H_3002__GP1      2021,349,23:59:59.999999 2021,350,23:59:59.998999  ==  1e+03 86400000

> wget -O mseed2 'https://service.iris.edu/ph5ws/dataselect/1/query?reqtype=FDSN&format=mseed&net=1H&sta=3002&cha=GP1&starttime=2021-12-22&endtime=2021-12-23&nodata=404'
> msi -tg mseed2
   Source                Start sample             End sample        Gap  Hz  Samples
1H_3002__GP1      2021,355,23:59:59.999999 2021,356,23:59:59.998999  ==  1e+03 86400000

PR #551 (2025-02-18): Make sure the beginning of the cut trace is at or after the cut time

### What does this PR do?
When the cut time starts in the middle of a sample the result will start slightly before the cut time.
This fix will make sure the beginning of the cut trace is at or after the cut time.

### Relevant Issues?
closes #550

### Checklist
- [ ] This PR is not directly related to an existing issue (which has no PR yet).
- [ ] All tests pass.
- [ ] Any new or changed features have are documented.
- [ ] Changes have been added to `CHANGELOG.txt` .
- [ ] First time contributors have added your name to `CON [... cut]
closed 2025-02-18 by timronan